chipsalliance/chisel #706: Cannot instantiate reset-free registers?

edwardcwang, 2017-10-13:
It seems that right now we cannot instantiate reset-free registers, which can be done in Verilog. This might be useful in things like synchronizers where we don't particularly care for reset, and somewhat limits the usefulness of UserModule/RawModule.

```scala
import chisel3._
import chisel3.core._
class Test extends UserModule {
  val io = IO(new Bundle{
    val in = Input(UInt(1.W))
    val out = Output(UInt(1.W))
  })
  io.out := RegNext(io.in)
}
```

```scala
chisel3.Driver.execute(Array[String](), () => new Test)
```
Results in an error since Reg's constructor depends on implicit clock and reset:

```
chisel3.internal.ChiselException: Error: No implicit clock and reset.
  at chisel3.internal.throwException$.apply(Error.scala:13)
  at chisel3.internal.Builder$.forcedClockAndReset(Builder.scala:205)
  at chisel3.internal.Builder$.forcedClock(Builder.scala:207)
  at chisel3.core.Reg$.apply(Reg.scala:23)
  at chisel3.core.RegNext$.apply(Reg.scala:41)
  ... 55 elided
```

jackkoenig, 2017-10-19:
This is related to a limitation in the current implementation of implicit clocks and resets where they are linked. Another way this bug comes up is if you try to do just `withClock` or just `withReset` in a RawModule. Shouldn't be too much effort to fix.

jackkoenig, 2020-05-19:
I'm not sure when this was fixed, but it is in `3.2.0` and `3.3.0` at least: https://scastie.scala-lang.org/U8Apga3XQ52jldHHsbxexA

closed 2020-05-19 by jackkoenig