chipsalliance/chisel #709: Unable to generate error due to sonatype/scala dependencies

DavideConficconi, 2017-10-19:
Hi, 
I'm using Chisel to create a design but I am experiencing an error. Until yesterday I was able to generate the Verilog but today I get this error : 

```

 [info] Updating {file:/home/dave/Documents/myfolder/}myfolder...
 [warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel3_2.11;3.0-SNAPSHOT
 [warn] Choosing sonatype-snapshots for edu.berkeley.cs#firrtl_2.11;1.0-SNAPSHOT
 [warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel-iotesters_2.11;1.1-SNAPSHOT
 [warn] Choosing sonatype-snapshots for edu.berkeley.cs#firrtl-interpreter_2.11;1.0-SNAPSHOT
 [info] Done updating.
 [warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
 [warn] 	* org.scalatest:scalatest_2.11:3.0.1 is selected over 2.2.5
 [warn] 	    +- edu.berkeley.cs:firrtl-interpreter_2.11:1.0-SNAPSHOT (depends on 3.0.1)
 [warn] 	    +- edu.berkeley.cs:chisel-iotesters_2.11:1.1-SNAPSHOT (depends on 3.0.1)
 [warn] 	    +- default:sdaccel_chisel_integration_2.11:1.0        (depends on 2.2.5)
 [warn] 	* org.scalacheck:scalacheck_2.11:1.13.4 is selected over 1.12.4
 [warn] 	    +- edu.berkeley.cs:firrtl-interpreter_2.11:1.0-SNAPSHOT (depends on 1.13.4)
 [warn] 	    +- edu.berkeley.cs:chisel-iotesters_2.11:1.1-SNAPSHOT (depends on 1.13.4)
 [warn] 	    +- default:sdaccel_chisel_integration_2.11:1.0        (depends on 1.12.4)
 [warn] Run 'evicted' to see detailed eviction warnings
 [info] Compiling 5 Scala sources to /home/dave/Documents/myfolder/target/scala-2.11/classes ...
 [error] /home/dave/Documents/myfolder/src/main/scala/package/MyClass.scala:39:22: value unary_! is not a member of chisel3.core.Reset
 [error]   slave_fsm.reset := !reset
 [error]                      ^
 [error] /home/dave/Documents/myfolder/src/main/scala/package/MyClass.scala:40:22: value unary_! is not a member of chisel3.core.Reset
 [error]   RTLKernel.reset := !reset
 [error]                      ^
 [error] two errors found
 [error] (compile:compileIncremental) Compilation failed
 [error] Total time: 15 s, completed Oct 19, 2017 12:39:29 PM
```

My build.sbt file is:

```
name := "chisel-module-template"
version := "1.0"
scalaVersion := "2.11.7"
parallelExecution := true
resolvers ++= Seq(
  Resolver.sonatypeRepo("snapshots"),
  Resolver.sonatypeRepo("releases")
)
// Provide a managed dependency on X if -DXVersion="" is supplied on the command line.
val defaultVersions = Map(
  "chisel3" -> "3.0-SNAPSHOT",
  "chisel-iotesters" -> "1.1-SNAPSHOT"
  )
libraryDependencies ++= (Seq("chisel3","chisel-iotesters").map {
  dep: String => "edu.berkeley.cs" %% dep % sys.props.getOrElse(dep + "Version", defaultVersions(dep)) })
libraryDependencies ++= Seq(
  "org.scalatest" %% "scalatest" % "2.2.5",
  "org.scalacheck" %% "scalacheck" % "1.12.4")
```

Any suggestion? Does anyone knows how to solve it?

Thanks

ucbjrl, 2017-10-19:
As of release v3.0-SNAPSHOT_2017-09-14, Module `reset` is no longer a Bool. Although we do support a `toBool` cast, connections of `Bool`s to `reset`s require a little more work. I would suggest the following:
```scala
   slave_fsm.reset := (!reset.toBool).asTypeOf(reset)
   ...
   RTLKernel.reset := (!reset.toBool).asTypeOf(reset)

```
or, revert to release v3.0-SNAPSHOT_2017-08-16 by editing your `build.sbt` file:
```bash
// Provide a managed dependency on X if -DXVersion="" is supplied on the command line.
val defaultVersions = Map(
  "chisel3" -> "3.0-SNAPSHOT_2017-08-16",
  "chisel-iotesters" -> "1.1-SNAPSHOT_2017-08-16"
  )
```

DavideConficconi, 2017-10-19:
Thanks, in this way it works

ucbjrl, 2017-10-19:
I'm going to change the title of this issue to more accurately reflect the problem.

jackkoenig, 2017-10-19:
Is the `asTypeOf` cast necessary? I had no problem with a small example doing:
```scala
  mySubModule.reset := !reset.toBool
```
I recognize that by casting it back to it's type, it should ensure this connection works for non-Bool resets, but I'm not sure this should/will even be allowed for asynchronous reset at least.

Just to provide some context, we changed reset from type Bool to a new type Reset in anticipation of expanding Chisel to support asynchronous reset in the future. While connecting to the submodule's reset as you're doing is totally fine, the suggested way is to change the implicit reset at time of instantiating the submodule like so:
```scala
  val slave_fsm = withReset(!reset.toBool)(Module(new SlaveFSM))
```
The main reason why this is suggested has to do with the fact that the type of the reset for a given module is the same as the type of the implicit reset in the scope in which it is instantiated. In synchronous reset code like it seems you are using, this is irrelevant. In cases where synchronous and asynchronous reset are mixed in the future, this distinction matters because the connection would fail while using `withReset` will work.

jackkoenig, 2017-11-14:
It looks like this has been resolved, I'm going to close it.

closed 2017-11-14 by jackkoenig