chipsalliance/chisel #710: User unfriendly or misleading error message when creating Chisel modules outside of a Builder context

edwardcwang, 2017-10-19:
Se #704

edwardcwang, 2018-12-18:
Confirmed to have a better error message now:

```scala
java.lang.IllegalArgumentException: requirement failed: must be inside Builder context
  at scala.Predef$.require(Predef.scala:277)
  at chisel3.internal.Builder$.dynamicContext(Builder.scala:193)
  at chisel3.internal.Builder$.readyForModuleConstr(Builder.scala:232)
  at chisel3.core.BaseModule.<init>(Module.scala:125)
  at chisel3.core.UserModule.<init>(UserModule.scala:18)
  at chisel3.core.ImplicitModule.<init>(UserModule.scala:121)
  at chisel3.core.LegacyModule.<init>(UserModule.scala:146)
```

makslevental, 2021-05-26:
the current [docs](https://www.chisel-lang.org/chisel3/docs/resources/faqs.html#get-me-verilog) have

```
import chisel3.stage.ChiselStage
object VerilogMain extends App {
  (new ChiselStage).emitVerilog(new HelloWorld)
}
```

when i do a similar thing 

```
object Im2Col extends App {
  val qSize = 15
  val stride = 1
  val padding = true
  val fifo = true
  val dWidth = 16
  val throughput = 1
  val inputCycle = 1
  val imgSize = 32
  val inChannels = 1
  val kernelWidth = 3

  val im2col = new Im2Col(
    UInt(dWidth.W),
    math.pow(2, imgSize).toInt,
    inChannels,
    2 * kernelWidth + 1,
    qSize,
    stride,
    padding,
    throughput,
    inputCycle,
    fifo = fifo
  )

  (new chisel3.stage.ChiselStage).emitVerilog(im2col)
}
```

i get this same exception

```
Exception in thread "main" java.lang.IllegalArgumentException: requirement failed: must be inside Builder context
	at scala.Predef$.require(Predef.scala:281)
	at chisel3.internal.Builder$.dynamicContext(Builder.scala:339)
	at chisel3.internal.Builder$.readyForModuleConstr(Builder.scala:499)
	at chisel3.experimental.BaseModule.<init>(Module.scala:168)
	at chisel3.RawModule.<init>(RawModule.scala:20)
	at chisel3.MultiIOModule.<init>(RawModule.scala:163)
	at chisel3.internal.LegacyModule.<init>(RawModule.scala:198)
	at systolic.Im2Col.<init>(Im2Col.scala:24)
	at systolic.Im2Col$.$anonfun$im2col$1(Im2Col.scala:173)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:52)
	at systolic.Im2Col$.delayedEndpoint$systolic$Im2Col$1(Im2Col.scala:163)
	at systolic.Im2Col$delayedInit$body.apply(Im2Col.scala:151)
	at scala.Function0.apply$mcV$sp(Function0.scala:39)
	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
	at scala.App.$anonfun$main$1$adapted(App.scala:80)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at scala.App.main(App.scala:80)
	at scala.App.main$(App.scala:78)
	at systolic.Im2Col$.main(Im2Col.scala:151)
	at systolic.Im2Col.main(Im2Col.scala)
```.

note that 

```
(new chisel3.stage.ChiselStage).emitVerilog(() => im2col)
```

gets me back to 

```
type mismatch;
 found   : () => systolic.Im2Col[chisel3.UInt]
 required: chisel3.RawModule
  (new chisel3.stage.ChiselStage).emitVerilog(() => im2col)
```