chipsalliance/chisel #712: About the write and read from Mem

zqcawzs, 2017-10-26:
The chapter Memories of chisel3 wiki shows that we can use mem.write and mem.read to write and read from a Mem. However, when I use this style in project, sbt shows compile error and I can only use mem(addr) := data style.

[error]   io.rdData := mem.read(io.rdAddr, io.ren)
[error]                        ^
[error] one error found
[error] (compile:compileIncremental) Compilation failed

edwardcwang, 2017-10-26:
Can you show more of the error message? Or a sample of your code?

zqcawzs, 2017-10-26:
Just a little example in chisel-tutorial, Memo.scala.
`when ( io.wen ) { mem.write( io.wrAddr, io.wrData) }
  .otherwise { io.rdData := mem.read( io.rdAddr, io.ren)}`
the solution:
`when (io.wen) { mem(io.wrAddr) := io.wrData }
   io.rdData := 0.U
   when (io.ren) { io.rdData := mem(io.rdAddr) }`
error:
[error] /home/zqc/chisel-tutorial/src/main/scala/problems/Memo.scala:27: too many arguments for method read: (idx: chisel3.core.UInt)(implicit compileOptions: chisel3.core.CompileOptions)chisel3.core.UInt
[error]   .otherwise { io.rdData := mem.read( io.rdAddr, io.ren)}
[error]                                     ^
[error] one error found
[error] (compile:compileIncremental) Compilation failed

sabusajin, 2017-10-26:
I think it's because mem.read only takes one argument.
Please refer-
https://chisel.eecs.berkeley.edu/api/#chisel3.core.Mem

zqcawzs, 2017-10-26:
But in the chisel3 wiki:
https://github.com/freechipsproject/chisel3/wiki/Memories

closed 2020-02-03 by albert-magyar