chipsalliance/chisel #716: printf %b displays true instead of bits

djg, 2017-11-06:
I'm trying to print the bits of a UInt(16.W) in groups of 4 with the following:

`printf("v: %b %b %b %b", io.in(15,12), io.in(11, 8), io.in(7,4), io.in(3,0))`

Instead of seeing the bit patterns, I see 

`v: true true true true`

colinschmidt, 2017-11-07:
What did you use to run this? Verilator? VCS? firrtl-interpreter? 

Does your emitted verilog have "%b"?

colinschmidt, 2017-11-07:
Based on some investigation by @jackkoenig it seems like you are probably using the firrtl-interpreter which does not convert from scala printf format specifiers to verilog format specifiers (or vice versa as the case may be).

@chick might be able to fix this.

djg, 2017-11-07:
Hi there. I just checked and I'm using firrtl by default.
I added a sample `printf(p"${Binary(io.io(15,12))}")` to my Module and I see the following in the .fir file:
```
    node _T_31 = bits(io.in, 15, 12) @[Shifter.scala 24:26]
    node _T_32 = bits(reset, 0, 0) @[Shifter.scala 24:9]
    node _T_34 = eq(_T_32, UInt<1>("h00")) @[Shifter.scala 24:9]
    when _T_34 : @[Shifter.scala 24:9]
      printf(clock, UInt<1>(1), "%b", _T_31) @[Shifter.scala 24:9]
      skip @[Shifter.scala 24:9]
```
so it looks like _T_31 which is the bits I want to print is being converted into a bool.

I'll convert to using verilator. Thanks.

chick, 2017-11-08:
This should be fixed with next release candidate in the firrtl interpreter

closed 2017-12-13 by jackkoenig