chipsalliance/chisel #717: Google Chisel Assert API

dt27182, 2017-11-07:
The assert API in chisel at google current looks like this:

object assert {
  def apply(
    cond: Bool, message: Printable = "", label: String, errorLevel: ErrorLevel = error
  )(implicit sourceInfo: SourceInfo): Unit = {
    ...
  }
  ...
}

The message is an optional Printable that Jack implemented. Not sure if he has upstreamed it. The label is a mandatory string. The ErrorLevel is an optional case object that is either "error" or "fatal".

The label is an required field because our DV infrastructure and best practices guide requires all asserts to have an label so that they can be easily switched on/off in the DV environment. I believe that this is a common industry practice. 

The other fields can live as additions to the API, but what do you guys think about making the label a mandatory field for all asserts?

ducky64, 2017-11-07:
What are the rules for the label? What happens if labels conflict? Can the label be automatically generated?
In software terms, I don't think I've seen labels in asserts before. Additionally, defaults and keyword arguments preclude the use of a macro as the function definition, though since you shouldn't be able to chain an apply after an assert, things like SourceInfoTransforms aren't needed

dt27182, 2017-11-08:
The label follows the same syntax rules as Verilog variables I believe. Labels are unique in verilog. We are using some auto name mangling to make sure labels don't conflict in Verilog, but we require the designer to manually put a meaningful label for every assert. This makes it possible for DV engineers to turn off a set of related asserts via regex. The emitted verilog looks like:

MY_ASSERT_LABEL: assert(...);

The dv people here say that it's common hardware verification practice to use labels for turning them on/off and for reporting purposes. Maybe the SiFive people can confirm.

@ducky64 Can you elaborate on the Scala software concern? This compiles and runs fine right now.

ducky64, 2017-11-08:
A limitation of the Scala macro system is that functions with either default or keyword arguments can't have a macro definition. We currently use macros for the source info system, where functions that take an implicit SourceInfo (or CompileOptions) are defined as macros which make the implicit explicit (tacking it onto the argument list) so there's no ambiguity if there's a chained apply (does it go into the implicit field? or is it a chained apply? the compiler usually doesn't do the right thing). I don't currently see a need for macros here (because you shouldn't need to chain an apply after an assert), but it would preclude that completely.

I think it would be good to have wider consensus about whether the label field should be required. @jackkoenig ?

azidar, 2017-11-21:
@jackkoenig and I agree that forcing users to label asserts is probably controversial, and merits a larger discussion. If we do decide that labelled asserts should be required, then there is a straightforward migration strategy of deprecating then erring on the old assert apply method.

In addition, this requires modifying FIRRTL to add an assert node, which is much larger change.
We feel this is a good use case to do a RFC proposal, and shouldn't old up the Chisel3 3.0 release.

dt27182, 2017-12-11:
@azidar Sounds, we can table this for later then.

closed 2017-12-11 by dt27182