chipsalliance/chisel #721: Strange behavior with .elements on anonymous bundle

jackkoenig, 2017-11-21:
It seems that if you pass an anonymous bundle to a function without first assigning it to a val, reflection fails to find anything so its elements are empty.

```scala
import chisel3._

trait HasFoo { def foo: UInt }
class TestBundle extends Bundle with HasFoo {
  val foo = UInt(32.W)
}

class TestModule extends Module {
  val io = IO(new Bundle { })
  def func(bun: Bundle): Unit = println(bun.elements)
  
  // This one has empty elements!
  func(new Bundle {
    val foo = UInt(32.W)
  })

  func(new TestBundle)
  func(new Bundle with HasFoo {
    val foo = UInt(32.W)
  })
  func({
    val b = new Bundle {
      val foo = UInt(32.W)
    }
    b
  })
}
```

This prints the following:
```scala
Map()
Map(foo -> chisel3.core.UInt@9)
Map(foo -> chisel3.core.UInt@b)
Map(foo -> chisel3.core.UInt@d)
```

edwardcwang, 2017-11-21:
Current differential diagnosis is a possible Scala reflection bug. Workaround: don't allow anonymous Bundles in this kind of scenario.

jackkoenig, 2017-11-21:
Possibly related:
```scala
class PrematureBundle extends Bundle {
  val foo = UInt(32.W)
  println(this.elements)
  val bar = UInt(32.W)
  println(this.elements)
}
```
By asking for elements before bar is declared, we just get:
```scala
Map(foo -> chisel3.core.UInt@7)
Map(foo -> chisel3.core.UInt@7)
```
But if you remove the first println you get:
```scala
Map(bar -> chisel3.core.UInt@8, foo -> chisel3.core.UInt@7)
```

jackkoenig, 2020-10-27:
This issue is still present, 3.4.0 Scastie: https://scastie.scala-lang.org/s2LvmYssTTO3BkWIxltNZQ

azidar, 2020-10-27:
Confirmed issue still exists in 3.4.0

jackkoenig, 2022-03-22:
What's bananas is that this is still present even in Chisel 3.5.2 when using the new `genBundleElements` support and Scala 2.13.8: https://scastie.scala-lang.org/ZOJTJTGGSXaO0xPoTKYzDw

No idea what is going on here...