chipsalliance/chisel #726: Provide API to Access When Condition

jackkoenig, 2017-12-07:
It would be very useful to be able to extract the when condition inside of a when block. 

```scala
when (...) {
  when (...) {
    // I want the current when predicate here
  }
}
```
One possible implementation is just a wire that defaults to false but gets true in the when block:
```scala
val pred = WireInit(false.B)
when (...) {
  when (...) {
    pred := true.B
    // use pred
  }
}
```

chick, 2017-12-08:
How about a `whenWithPredicate`
```scala
whenWithPredicate(pred) { pred1 =>
  whenWithPredicate(pred) { pred2 =>
     ???
  }
}
```
Could also be done with `elsewhen` too.

jackkoenig, 2017-12-08:
The primary use case is inside library functions where the library write wants to guard something with the and of the full when chain (so pred1 and pred2 aren't available). My particular use case is for an assertion/coverage library, I need the ability to guard stuff with the current when predicate. It would be incredibly onerous on the user for them to be required to keep track of the when condition themselves all over the place. It would also be nice for code to be able to at least inquire if it's currently nested in a when block since it might want to make it illegal to call from within a when.

jackkoenig, 2022-03-22:
Done via `when.cond` in 3.4.

closed 2022-03-22 by jackkoenig