chipsalliance/chisel #727: processing the UInt in bit

zqcawzs, 2017-12-07:
Hi, can I process UInt in bit? For example:

`val io = IO(new Bundle{
         val x = Input(UInt(10.W))
         val y = Output(UInt(10.W))
    })
    io.y(0) := io.x(0)`

I get error message like that when I intend to run these code:
`[error] (run-main-4) chisel3.internal.ChiselException: Cannot reassign to read-only chisel3.core.Bool@a
[error] chisel3.internal.ChiselException: Cannot reassign to read-only chisel3.core.Bool@a
`

jackkoenig, 2017-12-07:
Chisel3 does not support subword assignment. We find that this type of thing can usually be better expressed with aggregate types: Bundles and Vecs.

If you *must* express it this way, you can blast your UInt to a Vec of Bools and back
```scala
class TestModule extends Module {
  val io = IO(new Bundle {
    val in = Input(UInt(4.W))
    val bit = Input(Bool())
    val out = Output(UInt(4.W))
  })
  val bools = VecInit(io.in.toBools)
  bools(0) := io.bit
  io.out := bools.asUInt
}
```

edwardcwang, 2017-12-07:
It's on the Cookbook now: https://github.com/freechipsproject/chisel3/wiki/Cookbook#how-do-i-do-subword-assignment-assign-to-some-bits-in-a-uint

closed 2017-12-07 by edwardcwang