chipsalliance/chisel #728: DontCare Influences Type Inference

jackkoenig, 2017-12-08:
Consider the following:
```scala
class DontCareTest extends Module {
  class Bun extends Bundle {
    val foo = UInt(8.W)
  }
  val io = IO(new Bundle {
    val uint = Output(UInt())
    val bun = Output(Valid(new Bun))
  })
  val uint = WireInit(UInt(8.W), DontCare)
  io.uint := uint

  val bun = WireInit(Valid(new Bun), DontCare)
  bun.valid := false.B
  io.bun := bun
}
```

This gives the following error:
```
[info] Compiling 1 Scala source to /Users/jack.koenig3/ucb-bar/chisel3-raw/target/scala-2.12/classes...
[error] /Users/jack.koenig3/ucb-bar/chisel3-raw/src/main/scala/DontCare.scala:16: value valid is not a member of chisel3.core.Data
[error]   bun.valid := false.B
[error]       ^
[error] one error found
```

The definition of `WireInit.apply` being:
```scala
def apply[T <: Data](t: T, init: T)(implicit sourceInfo: SourceInfo, compileOptions: CompileOptions): T
```

Now changing the type of `init` to `Data` should fix this particular instance but this problem will pop up all over the place, including in libraries. Is there a way to tell the type system to ignore `DontCare` for type inference? @ducky64 @sdtwigg

ducky64, 2017-12-08:
What other instances of this problem are there? `:=` is already untyped, and changing `init` to type `Data` makes the `WireInit` equivalent to a `Wire` declaration followed by an untyped `:=` connect.

I'd imagine it doesn't make any sense to initialize a register with `DontCare`, since Chisel doesn't complain about uninitialized `Reg`s?

jackkoenig, 2017-12-08:
In library code if you accept both a `t` and an `init` it will do the same type inference and you'll get the same issue. That being said I don't think there's a fundamental fix other than "don't write bad library code".

PR #731 (2017-12-20): Add WireInit.apply that accepts DontCare
Prevents DontCare from affecting type inference
Fixes #728 

Just making `init` be of type `Data` would lose some static type safety, eg. `WireInit(new MyBundle, 0.U)` fails statically with this PR but wouldn't with changing the type of `init` to `Data`.
commit d95cb262 (2017-12-20): Add WireInit.apply that accepts DontCare (#731)

Prevents DontCare from affecting type inference
Fixes #728
closed 2017-12-20 by jackkoenig