chipsalliance/chisel #733: Unable to install Chisel commit 66d687d9c7af34e9d7a061feba309ec7a6dbd261

sam-falvo, 2017-12-11:
**Note: for support questions, you are strongly advised to use [Stack Overflow](https://stackoverflow.com/questions/tagged/chisel)**.

This repository's issues are reserved for feature requests and bug reports.

* **Type of issue**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Other enhancement

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **What is the current behavior?**  `git clone git@github.com:ucb-bar/chisel3 && cd chisel3 && sbt compile` yields, after many minutes, a dependency version conflict which results in the following output:

```
[0m[[0minfo[0m] [0mLoading project definition from /home/kc5tja/tmp/chistest/chisel3/project[0m
[0m[[0minfo[0m] [0mUpdating {file:/home/kc5tja/tmp/chistest/chisel3/project/}chisel3-build...[0m
[0m[[0minfo[0m] [0mResolving org.scalastyle#scalastyle-sbt-plugin;1.0.0 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.scalastyle#scalastyle_2.10;1.0.0 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.scala-lang#scala-library;2.10.6 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.scalariform#scalariform_2.10;0.2.0 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving com.typesafe#config;1.2.0 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.scoverage#sbt-scoverage;1.5.0 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.scoverage#scalac-scoverage-plugin_2.10;1.3.0 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving com.typesafe.sbt#sbt-ghpages;0.6.2 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving com.typesafe.sbt#sbt-git;0.9.3 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.eclipse.jgit#org.eclipse.jgit;4.5.0.201609210915-r ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving com.jcraft#jsch;0.1.53 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving com.googlecode.javaewah#JavaEWAH;0.7.9 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.apache.httpcomponents#httpclient;4.3.6 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.apache.httpcomponents#httpcore;4.3.3 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving commons-logging#commons-logging;1.1.3 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving commons-codec#commons-codec;1.6 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.slf4j#slf4j-api;1.7.2 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving com.typesafe.sbt#sbt-site;1.3.0 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving com.lightbend.paradox#paradox-theme-generic;0.2.9 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving ws.unfiltered#unfiltered-directives_2.10;0.9.1 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving ws.unfiltered#unfiltered_2.10;0.9.1 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving ws.unfiltered#unfiltered-util_2.10;0.9.1 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving commons-codec#commons-codec;1.10 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving ws.unfiltered#unfiltered-filter_2.10;0.9.1 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving ws.unfiltered#unfiltered-jetty_2.10;0.9.1 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.eclipse.jetty#jetty-webapp;9.2.21.v20170120 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.eclipse.jetty#jetty-xml;9.2.21.v20170120 ...[0m
[A
[2K[0m[[0minfo[0m] [0mResolving org.eclipse.jetty#jetty-util;9.2.21.v20170120 ...[0m
[... 332 lines cut ...]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615)
	at java.lang.Thread.run(Thread.java:745)
[0m[[31merror[0m] [0m(coreMacros/*:[31mupdate[0m) sbt.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found[0m
[0m[[31merror[0m] [0mTotal time: 1 s, completed Dec 10, 2017 7:59:51 PM[0m
```

  * **What is the expected behavior?**  It should work per the documentation on Github's README.md view.
  * **Please tell us about your environment:**
    - version: `commit 66d687d9c7af34e9d7a061feba309ec7a6dbd261`
    - OS: `Linux enif 3.19.0-84-generic #92-Ubuntu SMP Fri Mar 24 15:46:19 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

* **What is the use case for changing the behavior?**
Self-evident.

* **Impact**
  - [x] no functional change
  - [ ] API addition (no impact on existing code)
  - [ ] API modification
  - [ ] unknown

* **Development Phase**
  - [ ] request
  - [ ] proposal

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

sam-falvo, 2017-12-11:
Awww, crap.  I read the readme, and completely missed that.  :(  Wall of text gets quite blurry to me, apologies for poor vision on my part.  TIL: master != release branch.

I'll try again from a release branch.

edwardcwang, 2017-12-11:
If you just want to use Chisel (as opposed to develop Chisel), you can just use the https://github.com/ucb-bar/chisel-template

sam-falvo, 2017-12-11:
Thanks for the mention of Chisel-template.  I will look into that.  My goal is indeed to get started with writing my own cores as well as working with the tutorials.

That said, I checked out branch v3.0.0 and `sbt compile` thankfully it built.  However, "sbt test" has not produced even one succeeding test.  This is just one excerpt of the awesomely voluminous quantities of failure reports:

```
[info] [0.000] Elaborating design...
[info] [0.001] Done elaborating.
Total FIRRTL Compile Time: 4.5 ms
verilator --cc /home/kc5tja/tmp/chistest/chisel3/test_run_dir/LastAssignTester/201712102026165359375093090562671/LastAssignTes
ter.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace -O1 --top-module LastAssignTester +define+TOP_TYPE=VLastAssignTester
 +define+PRINTF_COND=!LastAssignTester.reset +define+STOP_COND=!LastAssignTester.reset -CFLAGS -Wno-undefined-bool-conversion 
-O1 -DTOP_TYPE=VLastAssignTester -DVL_USER_FINISH -include VLastAssignTester.h -Mdir /home/kc5tja/tmp/chistest/chisel3/test_ru
n_dir/LastAssignTester/201712102026165359375093090562671 --exe /home/kc5tja/tmp/chistest/chisel3/test_run_dir/LastAssignTester
/201712102026165359375093090562671/top.cpp
%Error: /home/kc5tja/tmp/chistest/chisel3/test_run_dir/LastAssignTester/201712102026165359375093090562671/LastAssignTester.v:4
9: syntax error, unexpected INTEGER NUMBER, expecting IDENTIFIER
%Error: /home/kc5tja/tmp/chistest/chisel3/test_run_dir/LastAssignTester/201712102026165359375093090562671/LastAssignTester.v:7
1: syntax error, unexpected INTEGER NUMBER, expecting IDENTIFIER
%Error: /home/kc5tja/tmp/chistest/chisel3/test_run_dir/LastAssignTester/201712102026165359375093090562671/LastAssignTester.v:9
3: syntax error, unexpected INTEGER NUMBER, expecting IDENTIFIER
%Error: Exiting due to 3 error(s)
%Error: Command Failed /usr/bin/verilator_bin --cc /home/kc5tja/tmp/chistest/chisel3/test_run_dir/LastAssignTester/20171210202
6165359375093090562671/LastAssignTester.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace -O1 --top-module LastAssignTeste
r '+define+TOP_TYPE=VLastAssignTester' '+define+PRINTF_COND=!LastAssignTester.reset' '+define+STOP_COND=!LastAssignTester.rese
t' -CFLAGS '-Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VLastAssignTester -DVL_USER_FINISH -include VLastAssignTester.h' -Md
ir /home/kc5tja/tmp/chistest/chisel3/test_run_dir/LastAssignTester/201712102026165359375093090562671 --exe /home/kc5tja/tmp/ch
istest/chisel3/test_run_dir/LastAssignTester/201712102026165359375093090562671/top.cpp
```

Hoping this can be of use for you.  :)

sam-falvo, 2017-12-11:
Hmm...stand by on this -- I might have too old a version of Verilator.  Will rebuild and update here shortly.

sam-falvo, 2017-12-11:
Confirmed; looks like everything is working.  Thanks for your guidance!!

closed 2017-12-11 by sam-falvo