chipsalliance/chisel #734: Problem in Queue - java.util.NoSuchElementException: key not found: ram_opt

jkorinth, 2017-12-11:
**Note: for support questions, you are strongly advised to use [Stack Overflow](https://stackoverflow.com/questions/tagged/chisel)**.

This repository's issues are reserved for feature requests and bug reports.

* **Type of issue**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Other enhancement

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**
Checkout the [demo project](https://github.com/jkorinth/chisel3-queue-problem) - run `sbt run`.

  * **What is the current behavior?**
Exception: `java.util.NoSuchElementException: key not found: ram_opt`
  * **What is the expected behavior?**
Success (same as for `sbt "run 1"`).

  * **Please tell us about your environment:**
    - version: `3.1-SNAPSHOT`
    - OS: `Linux endor 4.4.0-96-generic #119-Ubuntu SMP Tue Sep 12 14:59:54 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

* **What is the use case for changing the behavior?**
Optional elements in bundle definitions.

* **Impact**
  - [ ] no functional change
  - [ ] API addition (no impact on existing code)
  - [ ] API modification
  - [x] unknown

* **Development Phase**
  - [ ] request
  - [ ] proposal

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

grebe, 2017-12-24:
I think this is actually a firrtl bug, reported in freechipsproject/firrtl#666.

azidar, 2018-03-02:
Subsumed by freechipsproject/firrtl#666.

closed 2018-03-02 by azidar