chipsalliance/chisel #736: Negedge-triggered SRAMs

ccelio, 2017-12-13:
We currently have `SeqMem` as a primitive for posedge-triggered SRAM; I would like to get people's thoughts on adding a corresponding primitive for negedge-clock-triggered SRAM.

This request may require negedge registers too, since it is common to flop the output of SRAM immediately. Perhaps optionally rolling the output flops into the NegSeqMem will be decent approach? Example: `SeqMem(n, w, latency=1, negedge=true)`.

-------

* **Type of issue**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Other enhancement

* **What is the use case for changing the behavior?**

Very clever designs can shave off a pipeline stage by having some logic trigger on the negative clock edge. For example, accessing an SRAM that is indexed by a hash function:

Posedge logic pipeline:

    Cycle1: compute hash; send idx to SRAM
    Cycle2: access SRAM; flop output because no time is left after cycle-long access.
    Cycle3: do stuff with data

Negedge logic:

    Cycle1-hi: compute hash; send idx to SRAM
    Cycle1-lo: access SRAM; flop output after cycle-long access
    Cycle2-lo: do stuff with data

* **Impact**
  - [ ] no functional change
  - [x] API addition (no impact on existing code)
  - [ ] API modification
  - [x] unknown

* **Development Phase**
  - [x] request
  - [ ] proposal

* **Other information** 

Here is the current workflow to effect this behavior:

* create a module with asynchronous-read memory
* Pad out with a register to effect the 1-cycle latency
* Blackbox/manually drop in neg-edge SRAM when performing actual ASIC design

I'd like to believe we can model negedge triggered memories in a way that doesn't blow up our ecosystem, like chisel-testers or FIRRTL compiler passes, but still provide a comfortable path to optimal ASIC designs. The closer the Chisel-generated behavioral module is to what the blackboxed implementation would be, the lower the design risk will be.

edwardcwang, 2017-12-13:
What's wrong with just inverting the clock?

Having a latency parameter might be nice for technology mapping though I'm not sure often they're used.

aswaterman, 2017-12-13:
Inverting the clock could work given an annotation to inform the CAD tools that clk and ~clk are related clocks, and to emit posedge ~clk instead as negedge clk.  Perhaps surprisingly, the tools have trouble with this.

Adding NegSeqMem is a slippery slope, since we'll also need to add NegReg, NegQueue, ...; we should define a general solution.

seldridge, 2017-12-13:
I'm 100% onboard, but I think the related discussion is going to be around extending FIRRTL's IR to have some information about clock edges (and updating the Verilog emitter to consume this). Note: this would be a FIRRTL specification update.

aswaterman, 2017-12-13:
Yeah. I think expanding the Firrtl IR to handle negedge clocking is inevitable.