chipsalliance/chisel #740: [RFC] IntraModule Namespaces

jackkoenig, 2017-12-14:
Provide an API for scoping temporary and non-temporary names within modules.

Related to #535 

Proposed API:
```scala
// Prefix only temporaries
withTempPrefix("_COV") { ... }
// Prefix all identifiers
withPrefix("jacklib") { ... }
```
I'm not sure if the user should be expected to give the `_` prefix, or if we should mandate that for "temporaries". It's currently used by Firrtl as the heuristic for determining which name to delete when constant propagating.

* **Type of issue**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Other enhancement

* **What is the use case for changing the behavior?**

Improve internal naming stability. This is particularly relevant for libraries where we would like library code generation to have minimal impact on the names of everything else.

Potential use cases: Assertion coverage library, ECO support (it is absolutely critical to minimize impact on surrounding code for ECOs, I think Firrtl support is also necessary here).

* **Impact**
  - [ ] no functional change
  - [ ] API addition (no impact on existing code)
  - [x] API modification
  - [ ] unknown

IMO changes to name generation are API modifications, but increasing stability should be strictly a good thing.

* **Development Phase**
  - [ ] request
  - [x] proposal

mwachs5, 2017-12-14:
would these be nestable? And could you have access to existing prefix?

```
withPrefix("_COV") {
  ...
  withPrefix(existingPrefix + "jacklib")
}
```

jackkoenig, 2017-12-14:
I originally was thinking just an absolute override, but this is a great question. One the one hand, providing the existing prefix would let people nest if they want; on the other hand, if you're using other people's code and they just override it when you want them to prepend your prefix.

It could always prepend outer prefixes:
```scala
withPrefix("jackslib") {
  withPrefix("COV") {
    // all names get prefixed with jackslib_COV
  }
}

edwardcwang, 2017-12-14:
Why not something like

```scala
withPrefix("jackslib") {
  withPrefix("COV") {
    // all names get prefixed with jackslib_COV
  }
  withPrefix("foobar", absolute = true) {
    // all names prefixed with foobar_
  }
}
```

jackkoenig, 2017-12-14:
@edwardcwang I think that seems pretty reasonable

ducky64, 2017-12-15:
Two potential issues:
- it's inside a sub-scope, which means reflection-based naming won't be able to name contents. So this would depend on `@chiselName`
- the variables declared inside the scope won't be visible from outside, which might make it difficult to use (either everything inside would have to update existing wires, be a return value of the withPrefix function, or assigned to pre-declared vars)

Can you give an example of where this helps? Can the use case be addressed with improvements to one of the existing naming systems without requiring additional user code?

jackkoenig, 2017-12-15:
I think it's find to depend on `@chiselName`, I think we should encourage it. Additionally, I think we should look into dropping reflection and adopting something more like diplomacy's [ValName](https://github.com/freechipsproject/rocket-chip/blob/09160d0cd59cf74366541270a225985812277343/macros/src/main/scala/ValName.scala) macro.

The variable visibility thing can be a problem, but it's the same problem as with `withClockAndReset`. That being said, this API is primarily geared toward libraries, in which case the vals are in functions anyway.

I gave [brief] examples in the original message but they are admittedly buried in "What is the use case for changing the behavior?" I'll expand on them a little bit.

The basic idea is people want to write libraries that will inject code but they don't want it to cause it to change the names of everything else in the RTL. Anything simulation (as opposed to RTL implementation), fits the bill. The most important use case; however, is ECOs. Chisel could support this natively which would potentially solve the biggest use case, but it's really critical that ECOs don't impact any of the rest of the code in the module.

ccelio, 2018-09-09:
I'd just like to add my extra +1 support for the usefulness of this API.

seldridge, 2018-09-17:
@jackkoenig: I think your solution of creating modules for the defined scopes and then inlining is the way to go. A module is a scope (with IO). This proposed API is also just a scope (with no IO). Inlining is safe/sane name mangling and IO -> wire conversion.

I do need to get `InlineInstances` to do the right thing, but I'm 95% sure that all the machinery already exists inside of `Uniquify`. Or: what I expected would need to be implemented in `InlineInstances` is already the way that `Uniquify` works... (I kept thinking, ["If I just had all the module internals in a fake bundle, this would be easy."](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/Uniquify.scala#L199) :+1:) 

Let me know if that makes sense.

Any issues foreseen with doing module wrapping of what's in the scope and inferring module IO that crosses the scope boundary?

@mwachs, @edwardcwang: For this proposed solution, nesting would happen automatically and there would be no way to turn this off (that I can think of). If you wanted to use the `absolute` option this would likely need to be implemented on the Chisel side or this would be disallowed and the user should rewrite their code if they don't want name nesting. Access to the existing prefix would be available (because it's in scope), but you (likely) wouldn't have much reason to use it. Disabling of inlining would be relatively easy to do if the user wanted that.

jackkoenig, 2018-09-19:
@seldridge Yeah I've become more convinced that scoping in a module and then inlining it + your changes to `InlineInstances` is the right way to go. If only we could treat full modules as a Bundle 😉 

When it comes to inferring module IO, if we exposed an API to be able to determine what module a given signal is bound to (ie. declared within), then it should be super easy to do and this could probably even be expressed entirely as a util/library. Otherwise, I'm not sure how else to infer ports *at Chisel elaboration* time with the current API.

jackkoenig, 2020-08-13:
Fixed by https://github.com/freechipsproject/chisel3/pull/1448

Use `chisel3.experimental.prefix` for the explicit API, most cases will be handled by implicit prefixing from the names of values.

closed 2020-08-13 by jackkoenig