chipsalliance/chisel #756: Where is the release branch?

poweihuang17, 2018-01-13:
In the README:
"Firrtl, which compiles Chisel's IR down to Verilog. If you're building from the release branch of chisel3, separate installation of Firrtl is no longer required: the required jar will be automatically downloaded by sbt. If you're building chisel3 from the master branch, you'll need to follow the directions on the firrtl project to publish a local copy of the required jar."
But there isn't a release branch in this repo now. Where could I find it?
Thanks in advance.

edwardcwang, 2018-01-13:
There are now release branches for every major version.

https://github.com/freechipsproject/chisel3/tree/3.0.x

We should probably update the README.

azidar, 2018-01-18:
@ucbjrl can you update the readme, then close this issue?

ucbjrl, 2018-01-18:
Addressed by #759

closed 2018-01-18 by ucbjrl