chipsalliance/chisel #760: SyncReadMems read with enable doesn't work in chisel3

jackkoenig, 2018-01-20:
```scala
import chisel3._

class MyModule extends Module {
  val io = IO(new Bundle {
    val addr = Input(UInt(3.W))
    val en = Input(Bool())
    val out = Output(UInt())
  })

  val mem = SyncReadMem(8, UInt(32.W))
  io.out := mem.read(io.addr, io.en)
}

object Test extends App {
  chisel3.Driver.execute(args, () => new MyModule)
}
```

Compiling the above code results in a `RefNotInitialized` exception in Firrtl.

* **Type of issue**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Other enhancement

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**
  Run my code above
  * **What is the current behavior?**
  RefNotInitialized Exception in Firrtl
  * **What is the expected behavior?**
  No exception

* **Impact**
  - [x] no functional change
  - [ ] API addition (no impact on existing code)
  - [ ] API modification
  - [ ] unknown

* **Development Phase**
  - [x] request
  - [ ] proposal

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

Below is (I think) a functionally equivalent workaround and effectively what we should be emitting here.
```scala
  io.out := DontCare
  when (io.en) {
    io.out := mem.read(io.addr)
  }
```

aswaterman, 2018-01-20:
Wouldn't that assign ```io.out``` the value of the memory's output port the cycle the read is initiated, not the cycle the read completes?

Regardless, ```io.out := mem.read(io.addr)``` is also a functionally correct workaround.

jackkoenig, 2018-01-20:
You're right, would the equivalent then actually be something like?
```scala
val addr = WireInit(UInt(3.W), DontCare)
when (io.en) {
  addr := io.addr
}
io.out := mem.read(addr)
```

edcote, 2018-04-14:
Does this happen to be related to the the issue I brought up in the users group:

https://groups.google.com/forum/#!topic/chisel-users/d-outotUNw8

?