chipsalliance/chisel #764: type Reset matches all Bools

donggyukim, 2018-01-21:
**Note: for support questions, you are strongly advised to use [Stack Overflow](https://stackoverflow.com/questions/tagged/chisel)**.

This repository's issues are reserved for feature requests and bug reports.

* **Type of issue**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Other enhancement

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**
  * **What is the current behavior?**

I want to figure resets out from type matches like:
```scala
chiselType match {
  case r: Reset =>
  ...
}
```
But it matches all `Bool`s as `Bool` is `Reset`:
https://github.com/freechipsproject/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/Bits.scala#L733

  * **What is the expected behavior?**

Of course, it should match `Reset`s only

  * **Please tell us about your environment:**
    Berkeley `a` machies.

* **What is the use case for changing the behavior?**

* **Impact**
  - [x] no functional change
  - [ ] API addition (no impact on existing code)
  - [ ] API modification
  - [ ] unknown

* **Development Phase**
  - [x] request
  - [ ] proposal

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

jackkoenig, 2018-01-23:
I'm a little confused here. `Bool` is a subclass of `Reset` so it should match.

donggyukim, 2018-01-23:
Yeah, what I'm saying is it shouldn't match(`Bool` shouldn't be (a subclass of) `Reset`) as there's no other way to inspect resets otherwise. Seems like it's also conceptually not correct.

donggyukim, 2018-01-23:
Ah, it's not a bug report, so it changed it

grebe, 2018-01-24:
What is the reason that `Bool` subtypes `Reset`? It seems to me naively that it should be the other way around, or perhaps that `Bool` should subtype `ToBoolable`. Not every `Bool` is a `Reset`, but every `Bool` is `ToBoolable` (and so is every `Reset`).

Side note for a personal vendetta: it makes me very queasy that `Bool extends UInt(1.W, lit)`. I suspect anything that relies on this (rather than `extends Bits(1.W, lit)`) is a bug or trivially fixable. `+`, `-`, `*`, etc. should probably trigger deprecation warnings or something like that. `true.B * 3.U` is nonsensical.

ducky64, 2018-01-24:
Not every Bool is a Reset, but every Bool can be used as a Reset (or at least that's the intent here). How bad the edge cases are, and whether they can be addressed without completely destroying old code might be worth discussing though.

Fun side note: [the Bits(...) companion object actually creates UInts](https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/package.scala#L235). Whether that should be fixed is also another question. I think the case so far is that most agree Bits should be different from UInt, but there's no agreement on specifics.

azidar, 2018-03-02:
This is expected behavior.

closed 2018-03-02 by azidar