chipsalliance/chisel #775: Undeclared reference exceptions when using SyncReadMem

rimasaviz, 2018-02-11:
When generating verilog for a module with two SyncReadMem instances, I get an undeclared reference exception error that seems wrong (or at the very least inaccurate and counterintuitive):

[error] firrtl.passes.CheckHighForm$UndeclaredReferenceException:  @[memcrap.scala 26:12:@28.6]: [module CrapMem] Reference mem1 is not declared.
[error] firrtl.passes.CheckHighForm$UndeclaredReferenceException:  @[memcrap.scala 28:14:@31.8]: [module CrapMem] Reference mem1 is not declared.

See attached code and error log.  The lines of code the error message refers to are inside a when statement, the memories are instantiated just afterwards:

  when (doread) {
    rdaddr := rdaddr + 1.U ***
    when (rdaddr === 47.U) {
      rdaddr := 0.U ***
      rdsel  := !rdsel
      doread := false.B
    }
  }
  val mem0 = SyncReadMem(48, UInt(8.W))
  val mem1 = SyncReadMem(48, UInt(8.W))

Moving the SyncReadMem instantiations above the when statements makes the error go away. Is this the expected behavior?? Doesn't seem right.

[memcrap.scala.txt](https://github.com/freechipsproject/chisel3/files/1714207/memcrap.scala.txt)
[memcrap.log.txt](https://github.com/freechipsproject/chisel3/files/1714210/memcrap.log.txt)

rimasaviz, 2018-02-12:
Upon further investigation, it appears that while moving the SyncReadMem instantiations above the when statements gets rid of the error messages, the Verilog that gets generated is not correct (reads from one of the two memories don't work).  Here's the problematic verilog (from the `always @(posedge clock)` block at the end of the file).

```verilog
    if(mem0__T_46_en & mem0__T_46_mask) begin
      mem0[mem0__T_46_addr] <= mem0__T_46_data; // @[memcrap.scala 15:25:@13.4]
    end
    if (_GEN_22) begin
      mem0_mem0_dout_addr_pipe_0 <= rdaddr;
    end
    if(mem1__T_48_en & mem1__T_48_mask) begin
      mem1[mem1__T_48_addr] <= mem1__T_48_data; // @[memcrap.scala 16:25:@14.4]
    end
    if (_GEN_28) begin
      mem1_mem1_dout_addr_pipe_0 <= rdaddr;
    end
```
and here's the source of the problem (earlier in the file):
```verilog
wire  _GEN_22;
assign _GEN_22 = 1'h0;

wire  _GEN_28;
assign _GEN_28 = doread;
```
Using a single SyncReadMem seems to work OK.  The use case that brought this issue to light necessitates using a single 128 deep memory in place of 2 x 48 deep memories so some space is wasted (not an issue for me at the moment).  Here's the chisel code that results in incorrect verilog (but no error messages): 

```scala
package crap
import chisel3._
import chisel3.util._

class CrapMem extends Module {
  val io = IO( new Bundle {
    val din  = Flipped(ValidIO(UInt(8.W)))
    val dout = ValidIO(UInt(8.W))
  })
  val wraddr = RegInit(0.U(6.W))
  val wrsel  = RegInit(false.B)
  val rdaddr = RegInit(0.U(6.W))
  val rdsel  = RegInit(false.B)
  val doread = RegInit(false.B)
  val mem0 = SyncReadMem(48, UInt(8.W))
  val mem1 = SyncReadMem(48, UInt(8.W))
  val mem0_dout = mem0.read(rdaddr)
  val mem1_dout = mem1.read(rdaddr)

  when (io.din.valid) {
    wraddr := wraddr + 1.U
    when (wraddr === 47.U) {
      wraddr  := 0.U
      wrsel   := !wrsel
      doread  := true.B
    }
  }

  when (doread) {
    rdaddr := rdaddr + 1.U
    when (rdaddr === 47.U) {
      rdaddr := 0.U
      rdsel  := !rdsel
      doread := false.B
    }
  }

  when (io.din.valid && !wrsel) {
    mem0.write(wraddr, io.din.bits)
  }

  when (io.din.valid && wrsel) {
    mem1.write(wraddr, io.din.bits)
  }

  io.dout.bits  := Mux(rdsel, mem1_dout, mem0_dout)
  io.dout.valid := doread
}

object BuildCrapMem {
  def main(args: Array[String]): Unit = {
    chisel3.Driver.execute(Array("-X", "verilog", "-td", "./verilog"), () => new CrapMem )
  }
}
```

﻿

jackkoenig, 2018-02-16:
I think this is related to https://github.com/freechipsproject/firrtl/issues/727

commit 758eb0ba (2018-03-07): Fix SyncReadMem.read; add test

SyncReadMem.read with an enable signal currently only works in
compatibility mode, where Wires are implicitly initialized to
DontCare.  Fix by explicitly assigning DontCare to the Wire.

This might fix #775.
commit 24788503 (2018-03-07): Fix SyncReadMem.read; add test

SyncReadMem.read with an enable signal currently only works in
compatibility mode, where Wires are implicitly initialized to
DontCare.  Fix by explicitly assigning DontCare to the Wire.

This might fix #775.
commit a4aa3929 (2018-03-07): Fix SyncReadMem.read; add test (#796)

SyncReadMem.read with an enable signal currently only works in
compatibility mode, where Wires are implicitly initialized to
DontCare.  Fix by explicitly assigning DontCare to the Wire.

This might fix #775.
closed 2018-03-07 by aswaterman