chipsalliance/chisel #777: [RFC] Literal Types

ducky64, 2018-02-14:
## Motivation
Currently, literals have the same type as any other hardware node, and conversions back to Scala types can fail at runtime. Which hasn't been much of a problem, because needing to extract a literal is a rare case. However new testers (#725) proposes the uniform use of literals in interfacing with the simulator, because there's a lot of convenience functions (like binary constructors) that Scala types like Int/BigInt don't have, as well as the ability to use bundle literal constructors (whenever that gets implemented). However, without type differentiation, failures can be pushed to runtime, which could lead to unpredictable and hidden failures.

## Proposal
Create a `trait Literal` which literal types would extend in addition to the current `Data` (chisel type) hierarchy.

### External APIs
A sketch of the new (essentially parallel) type hierarchy would be:
- `BitsLit` extends Bits with Literal
  - `.asInt` extracts the contents as an unsigned (unless overridden) Scala Int. Runtime exception if the literal is too large to fit in a Int.
  - TODO: should there be `.asLong`?
  - `.asBigInt` extracts the contents as an unsigned (unless overridden) Scala BigInt
- `UIntLit` extends UInt with BitsLit
  - this provides no additional features compared to Bits
- `SIntLit` extends SInt with BitsLit
  - `.asInt`, `.asBigInt` will return signed instead
- `BoolLit` extends Bool with UIntLit
  - `.asBoolean` extracts the contents as a Scala Boolean
- `FixedPointLit` extends FixedPoint with BitsLit
  - `.asFloat` extracts the contents as a Scala float

TODO: Is there an advantage to type paramterize Literal by the chisel type, eg `trait Literal[T <: Data]` and `UIntLit` extends UInt with Literal[UInt]? 

### Internal APIs
Literal values would no longer be associated with the Chisel types (so UInt would no longer have a `lit` 
/ `litArg` variable), they would instead be stored in their literal classes. A deprecated shim would be included for compatibility.

### Primary driving example
- new testers
- any other uses?

## Projected impact 
Should minimally affect existing code, since literals still extend (and can be used in places that accept) their previous chisel data types. Unsafe literal extraction operations would now be deprecated.

## Related issues
Bundle literal constructors (#418) would be a separate issue but use this infrastructure. Annotation macros could be used to automatically generate the parallel literal hierarchy.

* **Type of issue**
  - [ ] Bug report
  - [ ] Feature request
  - [X] Other enhancement

* **What is the use case for changing the behavior?**
Typesafe literal extraction for #725 

* **Impact**
  - [ ] no functional change
  - [X] API addition (no impact on existing code)
  - [ ] API modification
  - [ ] unknown

* **Development Phase**
  - [X] request
  - [ ] proposal

@jackkoenig @chick

edwardcwang, 2018-02-14:
So would operations like `.U` return `UIntLit`?

mwachs5, 2018-02-14:
If I create a `val foo = Wire(my FooBundle().fromUInt(0))`, would I then be able to ask `foo` for it's litVal?

What would be useful to me is to be able to compose a Bundle from some constant values, and then figure out the total constant value of the Bundle. Specifically, this is useful for reset values and/or Read-Only registers whose contents are constant but determined only at Chisel runtime.

chick, 2018-02-14:
It seems to me that the real need here is the Bundle and Aggregate Literals.  Testers does just fine with  Ints or BigInts.  DspTools, as an example, uses Bundles for its synthetic types and that's what's hard to use in testers, it was not hard to implement doubles, but complex is hard.  If this is necessary infrastructure for aggregate literals that's good, but it seems that we create a plan for aggregates first and then design this so that it enables aggregate literals.

mwachs5, 2018-02-14:
Here is an example where I couldn't do what I wanted to do:

https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/devices/debug/Debug.scala#L731

```
    val HARTINFORdData = Wire (init = (new HARTINFOFields()).fromBits(0.U))
    HARTINFORdData.dataaccess  := true.B
    HARTINFORdData.datasize    := cfg.nAbstractDataWords.U
    HARTINFORdData.dataaddr    := DsbRegAddrs.DATA.U
    HARTINFORdData.nscratch    := cfg.nScratch.U

...
      (DMI_HARTINFO    << 2) -> Seq(RegField.r(32, HARTINFORdData.asUInt(), RegFieldDesc("dmi_hartinfo", "" /*, reset=Some(HARTINFORdData.litValue)*/))),
```

ducky64, 2018-02-14:
@edwardcwang Idea is that literal constructors like `.asUInt` or `.U` would return `UIntLit`

@mwachs5 Bundle literal constructors is a separate (but related) issue. An API there might be `FooBundle.fromULit(...)` or `FooBundle(true.B, 42.U, ...)`, which returns a FooBundleLit type with Literal binding (which doesn't need to be re-bound to a Wire). This parallel type could be automatically generated from the Bundle using annotation macros (though it's unclear currently what limitations might be).

@chick Testers is implemented fine with Ints / etc, but things are less great if you need binary constructors (which conveniently exist for Chisel literals; yes, you could write your own helper, but why reinvent the wheel...) or bundle literals (yes, not implemented yet, but that's part of the plan). To do typesafe bundle literals (as in, your bundle literal isn't just a map of string to int), you need a parallel type hierarchy anyways, so this shouldn't change the implementation complexity.

After a bit more thought, this proposal could break things. One unfortunately interaction is that `chiselTypeOf` would not do the right thing given a literal, since its type signature returns the type of the input... which would give you the Literal type right back. It might be possible to make it do the right thing with some kind of fancy implicit magic, though that needs more thought, and also whether it would be possible to apply in a distributed manner (because of user-defined types, primarily Bundles). Documentation on implicit scope search does give some cause for optimism: https://docs.scala-lang.org/tutorials/FAQ/finding-implicits.html

ducky64, 2018-02-14:
Also cloneType would probably break if called on a literal, since it has to return its own type. But it's a good thing we're trying to get rid of that...

ChiselTypeOf could be fine, since it's a top-level function and could take an implicit parameter list (to determine proper return type) since you're unlikely to chain anything after ChiselTypeOf.

Def macro magic could also help (for compatibility purposes) in both cases, though that's building up quite the complex stack.

grebe, 2018-02-15:
Agree with @chick. One use case to add to the list is constructors/factories- with something like `DspComplex` we want to do something different when you say `DspComplex(realLit, imagLit)` and `DspComplex(realVar, imagVar)`. Ideally, the first would also have type `Literal` and not have any wires.

ducky64, 2018-02-16:
Played around with this today, ran into problems, mainly with cloning and the differences between how this is handled with Bundles vs. other Chisel types.

The problem is that cloning returns `this.type`, so anything cloned must return an unbound version of its own type. Which doesn't work for literals, since it needs to returns its (chisel type) superclass, eg `UIntLiteral` should clone into `UInt`.

One (pretty clean) way to solve this would be with typeclasses, eg `UInt extends Cloneable[UInt]`, and `UIntLiteral extends Cloneable[UInt]` (and the `Bool extends UInt` is solved by making the type parameter covariant, `Cloneable[+T <: Data]`. Where this breaks down is with user-defined Bundles, where additional syntax would be required in each user-defined Bundle. So this would break all existing code, and isn't a solution.

This gets even worse with the Bundle literals proposal, where for example DecoupledLiteral (generated by a macro) would clone into Decoupled. The macro could eliminate the boilerplate (`with Cloneable[MyBundle]` and `MyBundleLiteral`), but everything would still need to coexist with non-macro Bundle code.

Ideas? Possibilities seem to be:
- Break all the Bundle code, requiring either a Cloneable mixin or macro. Basically a non-starter.
- Maintain a separate clone hierarchy for Bundles specifically. Basically need to double up every function that needs to clone into a normal version (using Cloneable[T]) and a Bundle version (using this.type). This would break any code that returns a clone of an input, so probably also a non-starter.
- Don't do typesafe literals. There could still be some improvements to extract literals from Data types, for example providing more convenient `getLitInt`, `getLitBigInt` methods that would assume the input is a literal (and error out of it's not). Obviously, not type-safe, so discourages the use of literals in non-hardware-construction applications. For new testers, the solutions would either be non-typesafe use of Chisel literals, or use of Scala types (which would be incompatible with Bundle literals).
- Some cleverness that I don't know about yet. I'd love a solution in this category if anyone has thoughts.

grebe, 2018-02-16:
What about using typeclasses via the implicits mechanism?

Something like
```
trait Clonable[T] {
  def cloneType(data: T): T
}

final class ClonableOps[T](lhs: T)(implicit ev: Clonable[T]) {
  def cloneType: T = ev.cloneType(lhs)
}

class BundleClonable[T <: Bundle] extends Clonable[T] {
  def cloneType(data: T): T = ??? // auto clone type?
}

class UIntClonable extends Clonable[UInt] { /* etc. etc. */ }

object implicits {
  implicit def cloneableOps[T : Clonable](lhs: T) = new ClonableOps(lhs)
  implicit def BundleClonableImpl[T <: Bundle] = new BundleClonable[T]
}
```

Then you don't have the inheritance problem, at least. I think you can override clonetype for a bundle by having an implicit with a closer type.

edwardcwang, 2018-02-16:
Related issues:

    Bundle literals #418
    Statically usable literals #417

edwardcwang, 2018-02-16:
Related issues:
- Bundle literals #418
- Statically usable literals #417

ducky64, 2018-02-16:
We discussed this at the dev meeting today, with several possible solutions.

With regards to cost-benefit analysis, we do not want to break massive amounts of existing code. Runtime checks (possibly in conjunction with something needing less boilerplate) would be preferable in those cases. But for completeness, I'll write up everything that was discussed.

## Most promising: Literals with implicit 'upcast' to Data subtype
**As in, this is what we're going to try.**

`class UIntLiteral(...) extends Literal` with `implicit def UIntLitToUInt(x: UIntLiteral): UInt`.
This class would be returned by current literal constructors, like `42.U`.

Crucially, `UIntLiteral` does NOT extend `Data` to force the use of the implicit in casting to `UInt`, so it can be used in any place where `UInt` is needed. The main limitation of implicits is that they don't chain, but I can't think of any use case where chaining is required. So this isn't expected to break any existing code.

However, this isn't clean from a type hierarchy point of view, `UIntLiteral` is **not** a subtype of `UInt`, it can only be implicitly converted to where an `UInt` is needed. This may have implications for match statements that don't go through a function where the implicit conversion would happen. I don't think that is a common case.

Note: this might also allow typesafe hardware/chisel types with compatibility. That's something totally different to explore at a much future date.

## Fallback: literals with implicit runtime checks
Inversion of the above solution, in case it doesn't work. Literal constructors still return a bare eg `UInt` type, but there would also be a `UIntLiteral` class with an implicit (unsafe) conversion from `UInt` to `UIntLiteral`. The implicit conversion would do a runtime check, so runtime errors are still possible.

Main advantage is less boilerplate (the `require` statements are implicit and automatic) and type-level definition of literals (self-documenting code; also provide convenience functions like `.asInt` or `.asBigInt`).

## Solutions that break user APIs
The main problem with `UIntLiteral` extending `UInt` is that `cloneType` needs to return `this.type`, but `UIntLiteral` needs to `cloneType` (or, more intuitively, `chiselTypeOf`) into plain `UInt`. There are several ideas here:

One, as @grebe suggests, is typeclasses with implicits, using a `Cloneable[+T <: Data]` mixin, and an implicit conversion for Bundles. However, this requires the modification of APIs that follow the pattern of WireInit:

`def WireInit[T <: Data](that: T): T`
(ones that are type parameterized on a `Data`, which returns the input type, functionally similar to `this.type` without the nastiness)

would have to be turned into:

`def WireInit[T <: Data](that: Cloneable[T]): T`

We expect there's a LOT of user code (eg `Queue(input: T)`) that follows this pattern, so while this appears to be a clean solution, this would also break lots of code and requires knowledge of the Cloneable API (largely a Chisel implementation detail) for library developers. However, breakages are expected to be noisy (will trigger compile time failures, you'll never get wrong RTL) and can be fixed easily once you know how.

A similar solution is to use method overloading to define different behaviors based on the input type:
`def WireInit[T <: Data](that: Cloneable[T]): T`
`def WireInit[T <: Bundle](that: T): T`
This suffers similar problems as above.

## Solutions requiring extensive boilerplate
One thought was to have a parallel Literal type hierarchy, where all Data operations would be re-implemented for this parallel type. Probably not a great idea, creates huge code bloat, which translates into maintenance burden. Similar to the first solution, but with a less clean implementation.

grebe, 2018-02-20:
Re: your most promising idea, should `UIntLiteral extend Literal` or `UIntLiteral extend Literal[UInt]`. I'm thinking about how you would write a polymorphic generator with typesafe literals, i.e.
```
class Butterfly[T <: Data](fixedTwiddleFactor: Literal[T]) extends Module { ... }
```
seems good. I'm not sure how you'd do that if `Literal` isn't templated.

I think `this.type` is kind of the core problem we're fighting here, right? I think it would be nice if we could move more of the bindings system to the type system eventually. `this.type` is going to be a problem for every other type of binding. Do you think the promising idea extends to other binding types?

ducky64, 2018-02-20:
Yeah, I think it has to be type parameterized, Literal[T] needs a method `toData: T` that returns the base chisel type.

I think generalizing it to bindings is more difficult, at least without sacrificing compatibility. The reason is that Literal conversions are mostly unidirectional (it really doesn't make sense to have a Data -> Literal implicit), while you would need both a WireLike -> Data and Data -> WireLike implicit so old and new APIs can interact. The problem with implicits is they don't chain (so we break anything else that needs an implicit) and you can't call an apply directly after (so anything that does indexing breaks).

ducky64, 2018-02-21:
Ok, so the most promising solution also doesn't work seamlessly, the problem is that implicits are leaky and imperfect, they can't handle type-paramterized conversions.
Example, it can't figure how to convert `Seq[UIntLiteral]` to fit into `Seq[UInt]`. This breaks code in chisel3 core.

ducky64, 2018-03-05:
Thanks for playing everyone, this is dead and pieces will be rolled into testers.

closed 2018-03-05 by ducky64