chipsalliance/chisel #779: Floating output ports

abdullahrazakhan91, 2018-02-20:
Is there any ways to have opens outputs ports in chisel. I cannot have any opened output ports. 

Thanks

edwardcwang, 2018-02-20:
By "opens outputs ports", do you mean like a Verilog `z`?

abdullahrazakhan91, 2018-02-20:
In VHDL. those port are not connected can be have opened value
example : EIGHT_BITADDER_INST : eight_bitadder port map (
    a => a,
    b => b, 
    f => f, 
    c => open, 
    o => overflow,
    z => open
    );

I mean to say if I have a port that is  not connected to anything how can I get in generated Verilog. 

When way could be to assign constant value and removed it from the verilog.

seldridge, 2018-02-20:
You may be looking for `donttouch`? See: [donttouch API](https://chisel.eecs.berkeley.edu/api/index.html#chisel3.core.dontTouch$)

That should prevent Firrtl's Dead Code Elimination (DCE) pass from (correctly) pulling out unused signals.

azidar, 2018-03-02:
You also need to assign the open outputs to be don't care https://chisel.eecs.berkeley.edu/api/index.html#chisel3.core.DontCare$. Please reopen if you need more clarification.

closed 2018-03-02 by azidar