chipsalliance/chisel #781: Auto-wrapping IOs of compatibility mode modules from chisel3._ modules fails

colinschmidt, 2018-02-21:
* **Type of issue**
**Bug report**

* **If the current behavior is a bug, please provide the steps to reproduce the problem:**
Instantiate a compatibility mode module with no `IO` wrapping inside a `chisel3._` module.
Then use that compatibliity mode module's IOs in some sort of logic.

  * **What is the current behavior?**
You get a `ExpectedHardwareException` when using the inner module's IOs.
  * **What is the expected behavior?**
Compatibility mode code shouldn't break, and so the inner module should have its io wrapped in `IO`
  * **Please tell us about your environment:**
    (examples)
    - version: `e27657118ff5915b96f8e3a467d464245fe09769`
    - OS: `Linux a2 4.4.0-97-generic #120-Ubuntu SMP Tue Sep 19 17:28:18 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

* **What is the use case for changing the behavior?**
Its a bug and this code shouldn't break when its enclosing module is changed.
Its difficult to slowly upgrade to `chisel3._` if that is the case

* **Impact**
**no functional change**

* **Development Phase**
**request**

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. Stack Overflow, gitter, etc)

If I manually wrap the io bundle in `IO` the issue is resolved so it seems to have something to do with the code that should autowrap compatibility modules io bundles.

azidar, 2018-03-02:
@ucbjrl could not reproduce, perhaps discuss in person?

azidar, 2020-10-27:
No reproducible case, and likely fixed. Please reopen if issue persists.

closed 2020-10-27 by azidar