chipsalliance/chisel #795: Verilator error on OSX

MrArna, 2018-03-05:
Hi, 

I got the following error while running "_sbt test_" on my OSX system

`clang++    TLP_sender-harness.o verilated.o verilated_vcd_c.o VTLP_sender__ALL.a    -o VTLP_sender -lm -lstdc++
clang++    Adder-harness.o verilated.o verilated_vcd_c.o VAdder__ALL.a    -o VAdder -lm -lstdc++
Undefined symbols for architecture x86_64:
  "vl_finish(char const*, int, char const*)", referenced from:
      VL_FINISH_MT(char const*, int, char const*) in verilated.o
Undefined symbols for architecture x86_64:
  "vl_finish(char const*, int, char const*)", referenced from:
      VL_FINISH_MT(char const*, int, char const*) in verilated.o
ld: symbol(s) not found for architecture x86_64
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [VTLP_sender] Error 1
make: *** [VAdder] Error 1`

My workaround consisted in modifying the following line in the _verilator.cpp_ file:
`#ifndef VL_USER_FINISH`
with this:
`#ifdef VL_USER_FINISH`

And everything worked well, but it's just a workaround.

azidar, 2018-03-16:
what's the version of verilator you are using?

matthewelse, 2018-04-14:
I'm having the same issue as this with Verilator v3.922 on mac OS, any ideas?

```bash
$ verilator --version
Verilator 3.922 2018-03-17 rev verilator_3_920-32-gdf3d1a4
```

azidar, 2018-04-20:
We cannot reproduce this using verilator version 3.922 on our macbooks. We will close this issue, and others are free to reopen if the same bug occurs for them. Glad you found a work around!

closed 2018-04-20 by azidar

chrinels, 2018-07-12:
Just wanted to let you know I encounter the same error while going through the chisel-tutorial I get the same error running. The earlier mentioned workaround solves it.
```
./run-examples.sh GCD --backend-name verilator
```
```
...
Undefined symbols for architecture x86_64:
  "vl_finish(char const*, int, char const*)", referenced from:
      VL_FINISH_MT(char const*, int, char const*) in verilated.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [VGCD] Error 1
```
System:
macOS High Sierra (10.13.5)

Verilator installed using brew
```
verilator --version
Verilator 3.924 2018-06-12 rev UNKNOWN_REV
```

jackkoenig, 2018-07-12:
I actually ran into this issue recently as well on my new Macbook Pro but haven't had time to debug.

Possibly related, chisel3 implements `vl_finish` in `src/main/resources/chisel3/top.cpp` and FIRRTL implements `vl_finish` in `src/test/resources/firrtl/testTop.cpp`. chisel-testers does not implement it which suggests it's either supposed to be using the default implementation in verilator or is using the one in chisel3.

jackkoenig, 2018-07-12:
I forgot to say that I **don't** see the issue in firrtl, chisel3, nor the chisel-testers, but I **do** see it in dsptools.

It's most likely do to the flag `-DVL_USER_FINISH` being passed to Verilator which indicates that you are providing your own implementation of `vl_user_finish`. This flag is provided by the `verilogToCpp` function in FIRRTL's `BackendCompilationUtilities` so if anything uses that but doesn't provide an implementation of `vl_user_finish`, one would expect exactly the error that people are seeing.

ucbjrl, 2018-07-12:
What I fail to understand is why this appears to be Verilator version sensitive. Or is it the Verilator build process on Macs that has an element of randomness (wrt `VL_USER_FINISH`)?

jackkoenig, 2018-07-12:
Are we sure that it's verilator version related or perhaps just MacOS environment related?

ucbjrl, 2018-07-12:
I think it's more likely to be related to the OSX environment. Typically we (at least I) don't do builds unless there's a version change which is probably why at first glance it appears version related.

jackkoenig, 2018-07-12:
Okay it does appear to be related to the version of verilator, with `Verilator 3.900 2017-01-15 rev verilator_3_890-15-ge6d7e7e` it works for me but when I upgrade to `Verilator 3.924 2018-06-12 rev UNKNOWN_REV` I get the same issue

edcote, 2018-08-14:
The error also appears outside of OSX.  I am taking a look.

ucbjrl, 2018-08-14:
I believe this is an issue with [chisel-testers](https://github.com/freechipsproject/chisel-testers), addressed by freechipsproject/chisel-testers#208.

edcote, 2018-08-14:
I came to the same conclusion.  Was writing my findings as you replied @ucbjrl.

I presume this could have been avoided by regressing chisel-testers (and chisel-tutorial) against chisel3/firrtl/rocketchip.

ucbjrl, 2018-08-14:
It would have been caught in our standard regressions tests if we used multiple Verilator versions (or at least the one we were recommending).

edcote, 2018-08-14:
3.900 works not 3.922 as recommended in README.md