chipsalliance/chisel #800: [RFC] define Module to be MultiIOModule? and unifying the Module type hierarchy

ducky64, 2018-03-16:
Currently, the type hierarchy for modules is:
(superclasses upwards)
`BaseModule` (block with ports, blackbox-type modules also extend this)
`chisel3.experimental.RawModule` = `chisel3.frontend.UserModule` (contains user RTL with no IOs defined by default)
`chisel3.experimentalMultiIOModule` = `chisel3.frontend.ImplicitModule` (defines IOs `clock` and `reset` with mechanism to automatically connect those by clock and reset scopes)
`chisel3.Module` = `chisel3.frontend.LegacyModule` (additionally defines IO `io`, and enforces no other IOs are defined)
(subclasses downwards)

(note: `chisel3.frontend._` is an internal name for these classes, `chisel3._` or `chisel3.experimental._` is the name exposed to the public API in `package.scala`)

So there's several points to this proposal:

## Does Module extending MultiIOModule make sense?
Currently, `chisel3.Module` is defined as a LegacyModule, which is actually a subclass of `chisel3.experimental.MultiIOModule`. However, IMO, a plain reading of the names would imply the opposite relationship - MultiIOModule **is-a** Module, not the other way around (unless you know the details of implementation).

Possible solutions are to change the naming so the relationship is more obvious, or:

## Does it make sense to redefine Module to MultiIOModule?
Since `chisel3.Module` **is-a** MultiIOModule, redefining `chisel3.Module` to be MultiIOModule should be safe. However, from way back when, there were also concerns about whether allowing multiple top-level IOs was a good idea. Now that MultiIOModule has been in experimental for a significant time, it might be worth discussing this again based on people's experiences.

A second question would be whether to keep LegacyModule at all, either exposed under a different name in `chisel3._` (maybe `chisel3.SingleIOModule`?) or in compatibility `Chisel._`. Are there any advantages to restricting modules to a single IO such that we shouldn't remove that feature?

## Can we unify internal and external (interface / API) names?
The idea here would be to rename the internal names to match the external names. I don't expect this to be too controversial, this will mostly affect implementation details.

**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

edwardcwang, 2018-03-24:
If you remove LegacyModule, doesn't that break backwards compatibility since most chisel3 modules are that sort of single-IO bundle module?

Also, I recall last time there was some concern about dropping single IO since it was a preferred style for some people. (Maybe this was discussed already at the recent dev meeting.)

ducky64, 2018-03-24:
This would be an API addition change, not an API breaking change, so no existing code should stop working.

I also remember some concerns about supporting multiple IO, but those were hypothetical. It would be interesting to see how many of them are actually problematic, and what the trade-offs are now that this is used in the wild. From there, we can hopefully have a discussion about whether that should be standard or opt-in behavior.

The people at the dev meeting were in favor of this, it made module inheritance easier.

edwardcwang, 2018-03-24:
So would `import chisel3._; class foo extends Module` still pull in the single IO module?

ducky64, 2018-03-24:
Proposal is to have it pull in the multi IO module (well maybe, depending on what people thing). But that's not expected to break code, since the single IO module is actually a subclass of the multi IO module with extra checks to enforce that only `io` is used.

edwardcwang, 2018-03-24:
How does it affect libraries which rely on `chisel3.module`? (e.g. midas may use the `io` field of a Module to get the I/Os)

ducky64, 2018-03-24:
Depends on how it's implemented.
There are several possibilities, we're keeping implicit clock and reset in any case. Whether we want to include at least `io` is an open question. But that should depend on how often certain features are used (and if they're being used in a good way).

From your example, it would appear that midas does not work well with multi IO modules. So then that goes back to a question of trade-offs, are the benefits of multi IO module worth it? Are there alternative APIs we could expose to allow midas to continue working? Are there other uses of this feature in the wild?

davidbiancolin, 2018-03-24:
This not a problem in midas as we use getPorts to inspect the IO of modules.

grebe, 2018-03-27:
This may not break source compatibility, but it will definitely break binary compatibility. It seems to me weird things could happen with sbt using versions like 3.0.+ unless this is done on a major version boundary.

jackkoenig, 2022-03-22:
This has been done in Chisel 3.5.0.

closed 2022-03-22 by jackkoenig