chipsalliance/chisel #803: DontCare does not support Clocks

colinschmidt, 2018-03-28:
**Type of issue**: bug report | feature request

**Impact**: API modification

**Development Phase**: request

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Trying to connect a clock output to DontCare results in the following exception.

`Caused by: chisel3.internal.ChiselException: cannot connect chisel3.core.Clock@1c9176 and chisel3.core.DontCare$@0`

I would expect this to be supported. Its easy to work around by converting to/from Bool.
Is there a rationale for not support Clock or is it an oversight?

**Please tell us about your environment:**
chisel3 commit:
```
commit 10751f57c595c899b4bcb81a6b0f78319ca66eb2
Merge: dc20aa4 9fd9433
Author: Jim Lawson <ucbjrl@berkeley.edu>
Date:   Sun Mar 4 09:30:31 2018 -0800

    Merge branch '3.1.x' into 3.1.0
```

azidar, 2018-04-20:
This is an oversight. Will fix.

edwardcwang, 2019-04-27:
Would like to second this. The workaround for `val deadClock = Wire(Clock()); deadClock := DontCare` is also a kludge:

```scala
      val w = Wire(Bool())
      w := DontCare
      val deadClock = Wire(Clock())
      deadClock := w.asClock
```

ducky64, 2019-04-27:
Potentially related to the general connect semantics discussion in #1056 ?