chipsalliance/chisel #812: verilog output reg statement cannot be expressed in chisel3.

wierton, 2018-04-22:
I know that in chisel3, output reg can be implemented by
```
val io = IO(Bundle { val output = Output(Bool()); })
val output = Reg(Bool());
io.output := output;
```
but I want to know if there is any direct support in chisel3 to it, I have tried `val output = Output(Reg(Bool()))`, it just give me an runtime error.

**Type of issue**: bug report | feature request | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

ducky64, 2018-04-22:
There is not - Bundle and IO definitions are strictly structural (think C structs), they cannot include any logic / hardware (including registers and literals). While it could allow nifty one-liners (like you proposed) in some cases, it could also cause confusion. For example:

```scala
val a = Wire(new Bundle { val a = Reg(Bool()) } )
a.a := ...  // is this a wire, or registered and delayed by a cycle?
```
and
```scala
val myReg = Reg(Bool())
myReg := ...
val io = IO(new Bundle { val a = myReg } )
io.a := ...  // what happens to myReg now?
```
and
```scala
class MyBundle extends Bundle {
  val a = Output(Reg(Bool()))
}
...
val io = IO(new MyBundle)
io.a := ...  // a register would be instantiated here, which might not be obvious to the user, since the definition is in MyBundle, which may be elsewhere in the code
```

If you can come up with a set of coherent and intuitive semantics for what you want to do, feel free to write a proposal! However, do keep in mind that language design must balance many factors, like conciseness vs. readability, and our current design philosophy leans more towards encouraging writing correct (bug-free) Chisel over lines of code. Because bugs in your RTL are really, really bad...

wierton, 2018-04-23:
@ducky64 , emm, if Output(Reg) will cause inconsistence in language layer, is it possible to add an simple API called Delayed or Synced? so that I can use following codes to implement output reg:
```
val io = IO(Bundle { val in = Input(Bool()); val output = Output(Bool()); })
val output = Delayed(io.output);
```
this will simplify many redundant codes if this module has many input and output and all output need to be delayed by a cycle.

ducky64, 2018-04-23:
If you just want a delay-and-assign in the same line of code, have you considered looking at RegNext (single cycle delay) or [ShiftRegister](https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/util/Reg.scala) (parameterized cycle delay) functions? For example, you could do
```scala
val io = IO(new Bundle { val in = Input(Bool()); val output = Output(Bool()); })
io.output := RegNext(theWireYouWantDelayed)
```

wierton, 2018-04-23:
@ducky64 , thanks, I have searched on https://chisel.eecs.berkeley.edu/api/index.html and failed to find what I want, maybe the keyword I searched is wrong.

closed 2018-04-23 by wierton