chipsalliance/chisel #813: Why is Reg(init=,next=) throwing deprecation warnings?

ccelio, 2018-04-22:
**Type of issue**: question

**Development Phase**: request

**What is the current behavior?**

Chisel3 throws a deprecation warning for the following code:

```
val slot_p1 = Reg(init = false.B, next = next_p1) 
```

The message:
```
[deprecated] issue_slot.scala:71 (12 calls): apply is deprecated: "Use Reg(t), RegNext(next, [init]) or RegInit([t], init) instead"
```

**What is the use case for changing the behavior?**

What is the rational behind making this deprecated? I find it far clearer than `RegNext(next_p1, false.B)`.

edwardcwang, 2018-04-22:
Richard can probably comment more, but IIRC the main reason was type safety/usability/avoid depending on `null`. The following slide is probably relevant:

[image]

ducky64, 2018-04-23:
The attached slide is more about why we have the hardware types vs. chisel data types separation.

The Reg deprecation I think is because:
1. Scala macros are a mess and can't support keyword arguments, so you wouldn't get source locators
2. The default null would require you to do some funny typecasting when calling Reg inside a type-parameterized function, so in some cases your Reg calls need to be `Reg(t = null.asInstanceOf[T], init=false.B, next=...)`. And the error message the Scala compiler gave was not at all intuitive.

Basically, Scala limitations mean we can't have the syntax we'd like to have.

ccelio, 2018-04-23:
Thanks for explaining the justification.

closed 2018-04-23 by ccelio