chipsalliance/chisel #828: Is it possible to add support for Xilinx and Intel fpga development?

sunxindx3906, 2018-06-05:
Is it possible to add support for Xilinx and Intel fpga development?

ducky64, 2018-06-05:
What do you want to see in terms of FPGA development?

Chisel generates Verilog, which can then be passed through FPGA synthesis toolchains. Chisel does not aim to replace synthesis or placer / router tools, which you will have to use external toolchains for.

Are you looking for getting-started examples (there's a few chisel projects that target Xilinx chips)? Hard macro wrappers? Something else?

sunxindx3906, 2018-06-05:
Mixed use of IP（rtl） and scale

ducky64, 2018-12-18:
Closing: not sure if there's anything needed to be done from a dev perspective. If you want to see some particular development, like hard blocks standard libraries or something, feel free to re-open this, or open another issue with more details.

closed 2018-12-18 by ducky64