chipsalliance/chisel #830: RawParam not escaped properly

ducky64, 2018-06-08:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
```scala
class SB_IO(pinType: String) extends ExtModule(Map(
    "PIN_TYPE" -> RawParam("6'b0110_01"),
...
```

produces this FIR file:

```
  extmodule SB_IO : 
    output PACKAGE_PIN : Analog<1>
    ...
    
    defname = SB_IO
    parameter PIN_TYPE = '6'b0110_01'
    parameter PULLUP = 0
    ...
```

and FIRRTL crashes with:

```
[info] Running org.calsol.databucket.Top
[info] [0.003] Elaborating design...
[info] [1.000] Done elaborating.
line 1508:36 token recognition error at: ''\n'
line 1508:28 mismatched input 'b0110_01' expecting NEWLINE
line 1522:36 token recognition error at: ''\n'
line 1522:28 mismatched input 'b1001_00' expecting NEWLINE
line 1536:36 token recognition error at: ''\n'
line 1536:28 mismatched input 'b0000_01' expecting NEWLINE
line 1550:36 token recognition error at: ''\n'
line 1550:28 mismatched input 'b0000_01' expecting NEWLINE
line 1564:36 token recognition error at: ''\n'
line 1564:28 mismatched input 'b0000_01' expecting NEWLINE
[error] (run-main-2c) firrtl.SyntaxErrorsException: 5 syntax error(s) detected
firrtl.SyntaxErrorsException: 5 syntax error(s) detected
        at firrtl.Parser$$anonfun$1.apply(Parser.scala:43)
        at firrtl.Parser$$anonfun$1.apply(Parser.scala:27)
        at firrtl.Utils$.time(Utils.scala:182)
        at firrtl.Parser$.parse(Parser.scala:27)
        at firrtl.Driver$$anonfun$getCircuit$1$$anonfun$apply$3.apply(Driver.scala:194)
        at firrtl.Driver$$anonfun$getCircuit$1$$anonfun$apply$3.apply(Driver.scala:171)
        at scala.Option.getOrElse(Option.scala:121)
        at firrtl.Driver$$anonfun$getCircuit$1.apply(Driver.scala:171)
        at firrtl.Driver$$anonfun$getCircuit$1.apply(Driver.scala:160)
        at scala.util.Try$.apply(Try.scala:192)
        at firrtl.Driver$.getCircuit(Driver.scala:160)
        at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:213)
        at firrtl.Driver$$anonfun$execute$1.apply(Driver.scala:210)
        at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:129)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at logger.Logger$.makeScope(Logger.scala:127)
        at firrtl.Driver$.execute(Driver.scala:210)
        at chisel3.Driver$.execute(Driver.scala:182)
        at chisel3.Driver$.execute(Driver.scala:202)
        at org.calsol.databucket.Top$.main(main.scala:306)
        at org.calsol.databucket.Top.main(main.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
```

Also, the Lattice toolchain (iCEcube2) inexplicably treats 
```
  SB_IO #(.PIN_TYPE(6'b0110_01), ...
```

from

```
  SB_IO #(.PIN_TYPE(25), ...
```

so IntParam doesn't work. But surprisingly, Yosys does do the right thing...

**What is the expected behavior?**

RawParam doesn't crash FIRRTL

**Please tell us about your environment:**

chisel3 (testers2) / firrtl / treadle latest master

**What is the use case for changing the behavior?**

giving me FPGA IOs that actually work as IOs

ducky64, 2018-12-18:
Interestingly, Chisel doesn't have the fix to generate the escape sequence, but FIRRTL still appears to parse it fine.
```
    defname = SB_IO
    parameter PIN_TYPE = '6'b1010_01'
    ...
```
generates into
```
SB_IO #(.PIN_TYPE(6'b1010_01), .PULLUP(0), .NEG_TRIGGER(0), .IO_STANDARD("SB_LVCMOS")) SB_IO
```

though for whatever reason now, adding the escape sequence in Chisel manually propagates it to FIRRTL and the escape sequence ends up in Verilog, which breaks things.

chick, 2020-03-30:
I have several places where it appears that people use
`"6\\'b0110_01"` to get around this. Should we close this issue or keep it open

ducky64, 2020-03-30:
We should probably define the behavior we want and add it to unit tests. Though the problem seems to be the current behavior is unexpected (strings aren't auto-escaped when generating FIRRTL) and changing to expected behavior might break existing code...