chipsalliance/chisel #834: Programmatic access to Module IOs

ducky64, 2018-06-09:
With MultiIOModule seeing a surprising amount of use, Module IOs are no longer limited to just `io`, `clock`, and `reset`. Use cases for programmatic access to IOs include in infrastructure (for example, testers), generators that pass through IOs, and bulk-connecting modules directly.

Rumor also has it that certain users in need of this feature have been punching through Chisel abstraction layers by defining a public method that passes through the results of `getModulePorts`.

## Proposal
Add a method `modulePorts(m: Module)` in `DataMirror` that returns the Module's top-level ports (including clock and reset, if applicable).

`DataMirror` indicates that a function is reflection-like (and is geared more towards infrastructure builders than RTL developers).

## Issues and Discussion
There are two options for the return type:
1. `Map[String, Data]`. The Data can be the Module port objects directly. But the map as a whole can't be directly used in Chisel.
2. `Record`. Probably more useful from a Chisel standpoint (and strictly more featured than a Map), but it's unclear what the binding would be, since elements of an aggregate (in this case, the module top-level ports) can't have a top-level (like module-port) binding.

**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

ducky64, 2018-06-15:
Notes from today's meeting discussions:
- Bulk connection capability is desirable, ideally where the result can be used on either side of the bulk connect operator.
- If we go with the Map option: it may be possible to do some kind of implicit conversion to BulkConnectable.
- While the ideal solution is tending towards Record, we can implement the Map version as a immediate experimental API if the bindings issue is too complex to deal with right now.

azidar, 2019-12-09:
Is already added.

closed 2019-12-09 by azidar