chipsalliance/chisel #836: Should we be able to poke() implicit resets?

edwardcwang, 2018-06-19:
While I'm aware that the testers have [reset()](https://github.com/freechipsproject/chisel-testers/blob/b1a69a77d561dec16a0161b56d85c1437e3ff89d/src/main/scala/chisel3/iotesters/ChiselPokeSpec.scala#L72), should we still be able to `poke()` reset like a normal signal? e.g. `poke(c.reset, 1)`

If so, it doesn't seem possible to do that yet since `poke()` doesn't recognize `chisel3.core.Reset`.

If not, then perhaps a more user-friendly error message is in order, since an error message of "no overload for `chisel3.core.Reset`" isn't immediately obvious - maybe we should add an explicit `poke()` method for `chisel3.core.Reset` and have it print an error message to use `reset()` instead.

jackkoenig, 2018-06-19:
I think you're going to want to be able to poke reset in some way, especially if you have multiple resets.

chick, 2018-06-20:
I am not aware of any reason we can't do this now

azidar, 2019-12-09:
@ducky64 reopen in testers2 and try with verilator.

closed 2019-12-09 by azidar

ducky64, 2019-12-09:
Looks like poking reset is already in the testers2 test suite: https://github.com/ucb-bar/chisel-testers2/blob/1760d3b134b59fba990c8b91bb2696c542d95dff/src/test/scala/chisel3/experimental/tests/VerilatorBasicTests.scala