chipsalliance/chisel #841: Return in When statement causes exciting bugs and error messages

mehnadnerd, 2018-06-29:
**Type of issue**: bug report/feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Code sample:
`
def safeAccess(): UInt = {
  when(cacheUsable) {
    return reqCache.data
  }
  return io.req.data.bits
}`
**What is the current behavior?**
Currently, this fails during elaboration, with the error message "Internal Error! When depth is != 0, this should not be possible".
This can be fixed with the below, under "Expected behaviour".

**What is the expected behavior?**
There are two possibilities: Either this is the wrong way to do it, and the correct way is:
`
def safeAccess(): UInt = {
  val temp = Wire(UInt())
  when(cacheUsable) {
    temp := reqCache.data
  } .otherwise {
    temp := io.data.bits
  }
  temp
}
`
or this should be allowed. (Is a Mux the correct option? This should compile to the same as one)
If this is the wrong way (which I think is likely), then it would be helpful to have a better error message and/or guidance that this is the wrong way to do it.

**Please tell us about your environment:**
a6

**What is the use case for changing the behavior?**
Easier use of functions for convenience.

edwardcwang, 2018-06-29:
Just for curiosity, what is the behaviour of the following?

```scala
def safeAccess(): UInt = {
  when(cacheUsable) {
    return reqCache.data
  } .otherwise {
    return io.req.data.bits
  }
}
```

Not sure on whether or not return in when() is intended to be allowed, but `Mux()` seems to be a good fit for your usecase.

mehnadnerd, 2018-06-29:
That won't compile, Scala doesn't recognise that the when ... .otherwise covers all cases and so doesn't know a UInt will be returned

mehnadnerd, 2018-06-29:
Results from Chisel-Dev Meeting:
put quotes around the ‘when’ in the error message, add a probable cause diagnosis, the correct way to do this is a mux, remove the ‘this isn’t possible’
return in when is not allowed, need to make it clear that this isn't the right way, use mux instead

PR #842 (2018-06-30): Catch returns from within when blocks and provide an error message
Resolves #841 

It turns out that non-local returns are an exception so we can catch this 💯 

**Related issue**:  #841 

**Type of change**: bug fix

**Impact**: no functional change (better error message)

**Development Phase**: implementation

**Release Notes**

Returns within when scopes used to give opaque error messages. Now they are caught earlier (with a stack trace) and explained in the error message.
commit b17998ea (2018-06-30): Catch returns from within when blocks and provide an error message (#842)

Resolves #841
commit 7ef67169 (2018-10-23): Catch returns from within when blocks and provide an error message (#842)

Resolves #841

(cherry picked from commit b17998eacb8e7b38b90829279e852bf8d5911f83)
closed 2018-06-30 by jackkoenig