chipsalliance/chisel #845: Trim Stack traces by default

jackkoenig, 2018-07-06:
**Type of issue**: feature request

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

Stack traces are great for helping users identify where a certain issue is; however, Scala programs tend toward large stack traces which raise the cognitive effort of determining which part of the stack trace is **useful**. Especially for people who are relatively new to Chisel and Scala.

Consider the following simple example:
```scala
import chisel3._

class NeedChiselType extends Module {
  val io = IO(new Bundle {
    val in = Input(UInt(8.W))
    val out = Output(UInt(8.W))
  })
  val delays = IndexedSeq.fill(4)(Reg(io.in))  // whoops, need a chiselTypeOf(...)
  delays(0) := io.in
  delays(1) := delays(0)
  delays(2) := delays(1)
  delays(3) := delays(2)
  io.out := delays(3)
}

object NeedChiselTypeTop extends App {
  chisel3.Driver.execute(args, () => new NeedChiselType)
}
```

**What is the current behavior?**

As expected, this gives a stack trace to point out our error!
```
[info] [0.001] Elaborating design...
[error] (run-main-5) chisel3.core.Binding$ExpectedChiselTypeException: reg type 'chisel3.core.UInt@e' must be a Chisel type, not hardware
[error] chisel3.core.Binding$ExpectedChiselTypeException: reg type 'chisel3.core.UInt@e' must be a Chisel type, not hardware
[error]         at chisel3.core.requireIsChiselType$.apply(Binding.scala:42)
[error]         at chisel3.core.Reg$.apply(Reg.scala:20)
[error]         at chisel3.package$Reg$.apply(package.scala:273)
[error]         at NeedChiselType$$anonfun$1.apply(NeedChiselType.scala:9)
[error]         at NeedChiselType$$anonfun$1.apply(NeedChiselType.scala:9)
[error]         at scala.collection.generic.GenTraversableFactory.fill(GenTraversableFactory.scala:90)
[error]         at NeedChiselType.<init>(NeedChiselType.scala:9)
[error]         at NeedChiselTypeTop$$anonfun$2.apply(NeedChiselType.scala:18)
[error]         at NeedChiselTypeTop$$anonfun$2.apply(NeedChiselType.scala:18)
[error]         at chisel3.core.Module$.do_apply(Module.scala:49)
[error]         at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error]         at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error]         at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:297)
[error]         at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:295)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at chisel3.internal.Builder$.build(Builder.scala:295)
[error]         at chisel3.Driver$.elaborate(Driver.scala:93)
[error]         at chisel3.Driver$.execute(Driver.scala:140)
[error]         at chisel3.Driver$.execute(Driver.scala:202)
[error]         at NeedChiselTypeTop$.delayedEndpoint$NeedChiselTypeTop$1(NeedChiselType.scala:18)
[error]         at NeedChiselTypeTop$delayedInit$body.apply(NeedChiselType.scala:17)
[error]         at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error]         at scala.App$class.main(App.scala:76)
[error]         at NeedChiselTypeTop$.main(NeedChiselType.scala:17)
[error]         at NeedChiselTypeTop.main(NeedChiselType.scala)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.lang.reflect.Method.invoke(Method.java:498)
[error]         at sbt.Run.invokeMain(Run.scala:93)
[error]         at sbt.Run.run0(Run.scala:87)
[error]         at sbt.Run.execute$1(Run.scala:65)
[error]         at sbt.Run.$anonfun$run$4(Run.scala:77)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error]         at sbt.TrapExit$App.run(TrapExit.scala:252)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error]         at sbt.Run$.executeTrapExit(Run.scala:124)
[error]         at sbt.Run.run(Run.scala:77)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1150)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1145)
[error]         at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at scala.util.Try$.apply(Try.scala:209)
[error]         at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:748)
```

**What is the ~expected~ desired behavior?**

That is a bit of a mouthful. If we could trim either side of the Stack trace to just give:
```
[error] (run-main-5) chisel3.core.Binding$ExpectedChiselTypeException: reg type 'chisel3.core.UInt@e' must be a Chisel type, not hardware
[error] chisel3.core.Binding$ExpectedChiselTypeException: reg type 'chisel3.core.UInt@e' must be a Chisel type, not hardware
[error]         ...
[error]         at NeedChiselType$$anonfun$1.apply(NeedChiselType.scala:9)
[error]         at NeedChiselType$$anonfun$1.apply(NeedChiselType.scala:9)
[error]         at scala.collection.generic.GenTraversableFactory.fill(GenTraversableFactory.scala:90)
[error]         at NeedChiselType.<init>(NeedChiselType.scala:9)
[error]         at NeedChiselTypeTop$$anonfun$2.apply(NeedChiselType.scala:18)
[error]         at NeedChiselTypeTop$$anonfun$2.apply(NeedChiselType.scala:18)
[error]         ...
[error]         (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)
```
I think it would be much better. My proposed technique for this trimming is as follows:

From the top of the stack, find the first function not in a blacklist of packages: `chisel3`, `scala`, `java`, `sun`, `sbt` (and any others we realize are being used for internal reasons).

We want to do a similar filtering from the bottom of the stack trace, but we need to be careful. User code in their "main" will show up before Chisel elaboration. So we could hackily trace till we hit the `chisel3.internal.Builder`, **or** we could just catch exceptions in the Builder and then trim that stack trace. If then then "rethrow" the exception, we're still going to end up with the whole of sbt internals in the stack trace. Rather, I think we should report the error ourselves and just return fail. It's unclear how this will work with people and custom flows but we can figure that out.

**What is the use case for changing the behavior?**

Better error reporting!

edwardcwang, 2018-07-06:
:+1: 

A related issue is caused by sbt arbitrarily/prematurely trimming stacktraces and hiding the actually useful part and users have to resort to [arcane sbt incantations](https://stackoverflow.com/questions/7237824/how-can-i-get-complete-stacktraces-for-exceptions-thrown-in-tests-when-using-sbt) to show the full trace so that **they are even able** to find the useful lines.

jackkoenig, 2018-07-06:
Yeah good point. I think we should be the masters of our error reporting and avoid letting our choice build tool influence them.

ucbjrl, 2018-07-06:
chisel2 tried to do this. Check out `printChiselStackTrace()` in [ChiselError.scala](https://github.com/ucb-bar/chisel2-deprecated/blob/master/src/main/scala/ChiselError.scala)

jackkoenig, 2018-07-06:
Good point, we should definitely draw inspiration from there

jackkoenig, 2018-07-19:
Alternative proposal to trimming: use a color scheme (and print the legend)

jackkoenig, 2018-12-18:
Fixed by #931

closed 2018-12-18 by jackkoenig