chipsalliance/chisel #861: Compatibility mode discrepancies - Input() and Output() definitions.

ucbjrl, 2018-07-18:
**Type of issue**: bug report

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information**
Martin Schoeberl ran into problems running Chisel2 code in compatibility mode on Chisel3. He was using `IO()`, `Input()`, and `Output()` which were added to Chisel2 in ucb-bar/chisel2-deprecated#734 in a (possibly) misguided attempt to ease conversion between Chisel2 and Chisel3 and minimize code differences.

 Now, `IO()` is available, but neither `Input()` nor `Output()` are defined in the compatibility package.
I think the solution is to simply add:
```scala
  val Input   = chisel3.core.Input
  val Output  = chisel3.core.Output
```
to the compatibility package, but this should be discussed in a developer meeting.

**What is the current behavior?**
Using
```scala
val A    = Input(UInt(width = n))
```
in Chisel2 code produces
```bash
[error] .../foo.scala:10: not found: value Input
[error]     val A    = Input(UInt(width = n))
[error]                ^
```
when the compatibility (`Chisel`) package is imported in Chisel3.

**What is the expected behavior?**
Find the definition of `Input()` or `Output()`.

**What is the use case for changing the behavior?**
Remove a confusing road block to Chisel2 -> Chisel3 conversion.

ducky64, 2018-07-19:
I think in general we might want to better define what the compatibility package should provide and what it should be used for. It's kind of been an inconsistent mix of "bare minimum to keep rocket-chip building under chisel3 infrastructure" and "provide some kind of transition path from chisel2 to chisel3". Overall, though, I don't think we've been backporting features to Chisel._, and it may (probably would?) be a significant burden to do so.

edwardcwang, 2018-07-19:
I mean, if Jim made a PR, unless it breaks rocket-chip or introduces other silent harm, it seems like it would be an easy win to help people migrate from Chisel2. It's not like we actively introduce the compatibility layer to new users anyway.

ducky64, 2018-07-19:
I'm all for this change, but I think there's a larger issue of "what is the compatibility layer" that might warrant discussion,

schoeberl, 2018-07-20:
I think, if it is not much work, this would make transition easier: one can add Chisel 3 must have and still compile it in Chisel 2, move on to compatibility in Chisel 3 with those changes, and then finally more or less just change packets to the real Chisel 3.

Cheers,
Martin

edwardcwang, 2018-07-20:
Since #862 is merged, perhaps we could open a new issue or just discuss at the dev meeting if want to talk about "what should be the compatibility layer"?

closed 2018-07-20 by edwardcwang