chipsalliance/chisel #864: Zipping Vecs Together

mehnadnerd, 2018-07-25:
**Type of issue**: feature request

**Impact**: API modification (Possible Impact on Existing Code)

**Development Phase**: request

**What is the use case for changing the behaviour?**

Currently, when you zip Vecs together, you end up with a Seq composed of tuples of the types.
This means that you cannot do things that require it to be a vec, like indexWhere. VecInit doesn't work because Tuple2, Tuple3... don't extend Data.

My workaround was to create a local bundle, and create a vec of that which had the other vecs tied to it - in effect manually zipping it together.

This could be made better. We could make a hardware tuple class, so that the syntax would be more like:
`VecInit(vec1.zip(vec2).map(_.toHardwareTuple)).indexWhere(elt => // some function on elt._1 and elt._2)` (Thanks Jack)
Even better is if we would overload the zip method, so `vec1 zip vec2` would yield a `Vec[HardwareTuple2[A,B]]`, and that could be reduced to `(vec1 zip vec2).indexWhere(elt => // some function on elt._1 and elt._2)`
That might be more difficult and/or run into conflicts.

jackkoenig, 2018-07-25:
I have a sketch of what HardwareTuples could look like:

```scala
import chisel3._
import collection.immutable.ListMap

class HardwareTuple2[T1 <: Data, T2 <: Data] private (a: T1, b: T2) extends Record {
  val _1: T1 = a.cloneType
  val _2: T2 = b.cloneType

  val elements = ListMap("_1" -> _1, "_2" -> _2)

  def cloneType: this.type = (new HardwareTuple2(_1, _2)).asInstanceOf[this.type]
}

object HardwareTuple2 {
  def apply[T1 <: Data, T2 <: Data](a: T1, b: T2): HardwareTuple2[T1, T2] = {
    val w = Wire(new HardwareTuple2(a, b))
    w._1 := a
    w._2 := b
    w
  }
  def apply[T1 <: Data, T2 <: Data](tuple: (T1, T2)): HardwareTuple2[T1, T2] =
    apply(tuple._1, tuple._2)
}

class IndexWhereModule extends Module {
  val io = IO(new Bundle {
    val foo = Input(Vec(4, UInt(8.W)))
    val bar = Input(Vec(4, UInt(8.W)))
    val out = Output(UInt())
  })
  val combined = VecInit(io.foo.zip(io.bar).map(HardwareTuple2(_)))
  io.out := combined.indexWhere(x => (x._1 + x._2) === 123.U)
}
```

Obviously the full Tuple2->Tuple23 (or whatever it goes to) would be a lot, but having at least a few would be helpful for stuff like this (anywhere you want to use a Tuple in place of Data)

jackkoenig, 2018-10-12:
It would be nice if this could also support stuff like:
```scala
val (a, b) = Mux(cond, (c, d), (e, f))
(a, b) := Mux(cond, (c, d), (e, f))
```

closed 2021-12-08 by jackkoenig