chipsalliance/chisel #871: Missing the `.W` suffix when constructing a UInt does not fail; returns wrong type?

ccelio, 2018-08-10:
If you forget the `.W` suffix in declaring a node (`0.U(sz)`) you do not get an error on the declaration, but rather, much later you get a bizarre message if you perform a reassignment to your scala variable to point to a new chisel node that consumes the previous node.

Here is the code:

```scala
 object Dummy                            
 {                                       
    private val sz = 2                   
    def apply(in: UInt): UInt =          
    {                                    
       var x = 0.U(sz) // broken     
       //var  x = 0.U(sz.W) // works  
        x = ( x << 1.U) | in           
        x 
    }                                    
 }                                       
```
**Error message:**
```
[info] Compiling 1 Scala source to /something/target/scala-2.11/classes ...
[error] /somethingsomething/.scala:430:26: type mismatch;
[error]  found   : chisel3.core.UInt
[error]  required: chisel3.core.Bool
[error]       x= (x << 1.U) | in
[error]                          ^
[error] one error found
```
**What is the expected behavior?**

I expect to receive an error message on not putting in a Width suffix on `sz`.

**Please tell us about your environment:**

Using e03e3f730250673ffa48c000caa369ed582467a2.

aswaterman, 2018-08-10:
Rewrite it and you'll see why it doesn't fail:

    val foo = 0.U // declare a UInt whose value is 0
    val bar = foo(sz) // extract bit sz from foo

ccelio, 2018-08-10:
Ha, amazing.  

I have never written code like this myself, but the Chisel cheat sheet recommends the `0.U(sz.W)` pattern, which seems... dangerous. Should this idiom be discouraged and the cheat sheet updated?

https://chisel.eecs.berkeley.edu/doc/chisel-cheatsheet3.pdf

edwardcwang, 2018-08-10:
IIRC the old (Chisel2) idiom was `UInt(0, width=sz)` which was type-unsafe because it depended on default arguments... Not sure how you would specify a UInt literal of a fixed width without using this idiom.

aswaterman, 2018-08-10:
I've fallen into the same pit, @ccelio.  It's not clear what to do about it, though, because extracting from literals is obviously supposed to work.

ccelio, 2018-08-10:
`0.asUInt(sz.W)` is the new idiom. It's verbose and it's ugly but it doesn't lead to errors.

This is fairly problematic, IMO, because it was only a pure accident that this particular scenario even throws an error.

Throws error:
```scala
   var x = 0.U(sz)
   x = x | in
```

Doesn't throw error, but not what the designer intended:

```scala
   var x = 0.U(sz)
   val y = x | in // compiles
  // val y: Bool = x | in // required to catch the designer error.
```

ccelio, 2018-08-10:
@aswaterman You're correct, I have no idea how to catch this, other than to never encourage people to write `0.U(sz.W)`. And possibly to have a better constructor.

aswaterman, 2018-08-10:
Or we could un-hide the Lit method, so you can write `UInt.Lit(0, sz)`, kind of like Chisel2.

ccelio, 2018-08-10:
Is there a reason `UInt(0, sz.W)` is not available? And puke if `.W` is not specified so it's clear the intent of the designer is not `UInt(sz, 0)`, avoiding the initial problems of the `UInt()` constructor?

edwardcwang, 2018-08-10:
With regards to `UInt(0, sz.W)` - the syntax is "supported" but deprecated : https://github.com/freechipsproject/chisel3/blob/4de6848ef746ca40945dc95a113e820bc7265cea/src/main/scala/chisel3/package.scala#L162

ducky64, 2018-08-23:
Consensus seemed to have been "there isn't a good solution (especially considering the amount of code that's using this style), so there's not much we can do about it, it also surprisingly hasn't come up too much"
(if this is wrong, feel free to re-open)

closed 2018-08-23 by ducky64