chipsalliance/chisel #878: sub field assignment

schoeberl, 2018-08-23:
**Type of issue**: bug report

**Impact**:  Chisel 2 code needs changes

**Development Phase**: request

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

assign subfields to a register (worked in Chisel 2):

```
  val shiftReg = RegInit(0x7ff.U)
...
      when(io.channel.valid) {
        shiftReg(0) := 0.U // start bit
        shiftReg(8, 1) := io.channel.data // data
        shiftReg(10, 9) := 3.U // two stop bits
```
workaround:
```
        shiftReg := Cat(Cat(3.U, io.channel.data), 0.U) // two stop bits, data, one start bit
```

**What is the current behavior?**

[error] (run-main-0) chisel3.internal.ChiselException: Cannot reassign to read-only chisel3.core.Bool@74
[error] chisel3.internal.ChiselException: Cannot reassign to read-only chisel3.core.Bool@74
[error] 	at chisel3.internal.throwException$.apply(Error.scala:13)
[error] 	at chisel3.core.Data.connect(Data.scala:296)
[error] 	at chisel3.core.Data.$colon$eq(Data.scala:365)
[error] 	at uart.Tx$$anonfun$1$$anonfun$apply$mcV$sp$2$$anonfun$apply$mcV$sp$3.apply$mcV$sp(Uart.scala:55)

**What is the expected behavior?**

Work like in Chisel 2 or have an explanation in the porting Chisel 2 to 3 document.

**Please tell us about your environment:**

Mac

edwardcwang, 2018-08-23:
https://github.com/freechipsproject/chisel3/wiki/Cookbook#how-do-i-do-subword-assignment-assign-to-some-bits-in-a-uint

chick, 2018-12-18:
Document in chisel2 to 3

edwardcwang, 2018-12-18:
Just added to chisel2-to-3 page on wiki.

closed 2018-12-18 by edwardcwang

juliusbaxter, 2019-01-01:
Is there any discussion on why subword assignment was removed from Chisel 3? It seems to me to be an oft-used feature, where the proposed workarounds are all less concise.

ducky64, 2019-01-01:
There's a bit more information in #244, but the overall rationale is while there may be loss of conciseness, we think clarity will be improved by using Vec or Bundle as appropriate. Bit indexing in many cases is also dependent on word width, as opposed to Vec indexing or Bundle element access.

Though if you have compelling uses, feel free to make a case for this to be a feature again.

Otherwise, perhaps we should start including the suggested alternative and underlying rationale in the error message...

juliusbaxter, 2019-01-02:
Agree vecs and bundles are nice but you don't always have them set up to access individual bits, so it's inconvenient.

I think the main argument against the removal of subword assignment is consistency. If I can get a subword from a value, I should also be able to set it. Inconsistencies like this are frustrating for the user.

jonassvedas, 2019-01-11:
> Agree vecs and bundles are nice but you don't always have them set up to access individual bits, so it's inconvenient.
>
> I think the main argument against the removal of subword assignment is [...]

Totally agree. I find Vecs or Budnles are good when you want to have nicely structured signals but converting a 256 bit UInt to a Vec just so you could assign seems quite frustrating . Also I found these types of bit blasted Vecs come with a lot of problems too:
* you cannot apply arithmetic operations to them
* they will appear as separate signals in vcd waves which is not readable
* in the chisel iotesters you need to poke every vector value separately

tdb-alcorn, 2019-05-19:
+1 I'm generating ultra-wide buses to get around the problem of many Vecs slowing down the Chisel compiler, and it would be great to be able to assign to subwords. Currently I'm working around this by using verilog black boxes to subword assign instead (very hacky)

schoeberl, 2022-05-20:
However, I still think that subword (individual bit) assignment should be possible in Chisel (like in V*). I am reopening this issue. If this is too hard to implement then we can close it again (maybe with an argument).