chipsalliance/chisel #879: I still got the error message "edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT: not found" after publishLocal firrtl

PAIWANG94, 2018-08-24:
**Type of issue**: bug report | feature request | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

ucbjrl, 2018-08-24:
I suspect your versions of firrtl and chisel3 are inconsistent. We recently switched the default Scala version for both from 2.11 to 2.12. Make sure you're on the master branch for both and do a `git pull` to ensure they're up-to-date.

Alternatively, you can:
```bash
sbt +publishLocal
```
which will publish all cross versions (both Scala 2.11 and Scala 2.12 currently).

closed 2018-09-14 by jackkoenig

qiner199, 2018-12-03:
I update the build.sbt file with  
resolvers ++= Seq(
    Resolver.sonatypeRepo("snapshots"),
    Resolver.sonatypeRepo("releases")
  )
worked successful !! you can have a try

jackkoenig, 2018-12-03:
I believe that changing the resolvers is grabbing the published snapshots, not using a locally published version.

If you want to publish locally and use that one, you should use `sbt +publishLocal`