chipsalliance/chisel #886: Request for a NotSynthesized annotation on a per signal basis.

ccelio, 2018-09-08:
**Type of issue**:  feature request

**Impact**: API addition (no impact on existing code) 

**Development Phase**: request 

**What is the use case for changing the behavior?**

There are useful signals to add for debug and verification that would be too expensive to synthesize.

**What is the expected behavior?**

I would like some way to annotate Chisel nodes/signals/registers as "NotSynthesized"/"DontSynthesize"/whatever. 

Chisel/firrtl currently uses the define macro SYNTHESIS to gate off unsynthesized logic. Regarding the final code generation, perhaps one of two things can happen in the Verilog:

1) Hide unsynthesized nodes behind a `ifndef SYNTHESIS` macro.  This is probably not possible since I guess every use of these signals would also have to be gated off or else it would throw undefined errors.

2) Throw an error if SYNTHESIS is set and NotSynthesized signals exist in the Verilog code. This is weird (does Verilog have require statements?), and would put the burden on the Chisel RTL user to disable/remove NotSynthesized code from their design once they want to take the design through synthesis, but it helps catch mistakes (esp. when using someone else's code).

This is a bit pie-in-the-sky so I'd be interested in hearing other people's thoughts on this.

aswaterman, 2018-09-08:
Dead-code elimination in synthesis tools is very reliable.  Have you actually encountered a case where adding a bunch of debug-only registers and logic has affected QoR?

ccelio, 2018-09-08:
A couple of scenarios I'm interested in:

1) Hypothetically, some optimizations (like cross-module optimizations) may be verboten which prevents DCE from working its beautiful magic. Yah yah I know...

2) I've seen one case where DCE struggled to deal with Mem(new MyBundle) scenarios where most elements within MyBundle are unused, and I was expecting Chisel/Firrtl's DCE to have cleaned it up for me.  I need more time to delve into this particular issue, but at a meta-level, it would be nice to be able to have an annotation to sanity-check that nothing made it past firrtl's dce that I wasn't intending to. 

3) Going further on (2), in the short term, I'll likely hack together some annotations/artifacts to construct a file of unsynthesized signals so I can search synthesis reports to verify the synthesis DCE actually took place. But it feels like we can do better than just emitting a file of DontSynthesize signals that needs to be consumed by the build system (that requires awareness on the build system builder to know about this).

4) RTL designers accidentally hook their own logic off of unsynthesized debug signals and end up making them synthesizable. No bueno.

5) The DontSynthesize signals may very well be architecturally visible constructs that won't get DCE'ed (see Jerry's trace support for BOOM: https://github.com/ucb-bar/riscv-boom/pull/112).