chipsalliance/chisel #889: VCD Dump fails for FIR interpreter

tampler, 2018-09-14:
Bug report

I checked out the chisel-template and launched the default GCD Test with the following:
sbt:proj> test:runMain gcd.GCDMain --fint-write-vcd --is-verbose

This is supposed to dump a VCD file for a default FIR interpreter. This doesn't work now, no VCD is dumped. However, if I specify verilator backend with option "--backend-name verilator", both Verilog and VCD files are generated.

SBT 2.12.6, latest Chisel 3.1+

chick, 2018-09-14:
This problem was inadvertently caused by the a recent change that made treadle the default backend instead of the firrtl-interpreter. I think we should review that decision in light of this problem.

The immediate fix is to add command options to set the backend explicitly. This can be done in this case by changing line 101 of GCDUnitTest.scala to be 
```scala
    iotesters.Driver.execute(Array("--fint-write-vcd", "--backend-name", "firrtl"), () => new GCD) {
```

tampler, 2018-09-15:
Hi Chick

Okay, your workaround works with the explicit backend specification :
sbt 'test:runMain gcd.GCDMain --fint-write-vcd --backend-name firrtl'

Is this a bug? This adds some mess for newcomers. So either return this to the state it used to be or update a documentation.

Should I close this issue or leave it unresolved for some time ?

chick, 2018-09-15:
Please leave it open for now, we are discussing the solution. We'd rather have one flag but there's some complications. Thanks for bringing this up

chick, 2018-09-20:
The master version of chisel template now uses --generator-vcd-output on|off which works consistently across verilator, treadle and the interpreter.  I think this is a sufficient fix. It will join the general release soon

closed 2018-09-20 by chick