chipsalliance/chisel #891: Allow Referential Equality for Expressions

mehnadnerd, 2018-09-15:
**Type of issue**: feature request

**Impact**: API modification

**Development Phase**: request

**What is the current behavior?**
Currently, the same statement, written twice, will yield two results that are not referentially equal. IE, `a && b` and `a && b` are not referentially equal. This leads to problems such as https://github.com/freechipsproject/rocket-chip/issues/1616, where referential equality fails to identify the two expressions as the same.

**What is the expected behavior?**
They will compare equal.

**What is the use case for changing the behavior?**

This issue could also be addressed by improving the instance equality checks, so that referential equality could be avoided. However, the full implementation of this would be extremely difficult, as it would require canonicalising the boolean expressions.

jackkoenig, 2018-09-17:
I see a few ways this could be accomplished:
1. Aggressive memoization of intermediate results in Chisel
    * Pros: Would accomplish the results
    * Wash: Moderate code impact
    * Cons: Difficult and bug prone to implement, potentially large memory impact
1. Support structural equality and complex expressions in Chisel
    * Pros: Robust and immutable*, huge increase in information available at Chisel elaboration time
    * Cons: Huge code impact, removes much of the "lean frontend" aspect of Chisel
1. Rely upon FIRRTL to optimize
    * Pros: No impact on Chisel
    * Wash: Unclear impact on FIRRTL (*might* be relatively easy and contained
    * Cons: Requires emitting specific logic patterns

I think (3) is by far the most tractable and is somewhat orthogonal from (1) and (2) in that it has benefits for other code while (1) and (2) have their own additional benefits with regards to what information is available in Chisel.

mehnadnerd, 2018-09-19:
The issue I see with 1) is that currently case classes are used. We would basically have to rewrite case classes to support such memoisation (Actually, this would be a cool feature in Scala itself, but WAY beyond scope).

If we implement a very limited version of 2), i.e. only supporting expressions that are exactly identical `a && b == a && b` but `a && b != b && a` (Note Scala operators), it could be easier. SourceInfo is the main issue preventing this limited version of structural equality, I'm not sure if there is an easy way to exclude it.
It is possible it might be easier to add some other operator that simply checks if they are equal (Possibly recursively), ignoring SourceInfo, which could be useful in this and other contexts. Making them compare as equal when strictly speaking, they aren't, might or might not be a bad idea.

As for 3), are we planning on implementing logic optimisation in FIRRTL? I was under the impression that we relied on the backend to do that, and the issue with the solution of switching to a logically equivalent but more complex expression would be that not all backends can be counted on to do that.

mehnadnerd, 2018-09-21:
Closing, Chisel meeting 2018-09-21 decided that would be not a good idea to add to Chisel, the warning in RocketChip should serve as enough.

closed 2018-09-21 by mehnadnerd