chipsalliance/chisel #897: sbt complains about feature warnings which are irrelevant to new users

edwardcwang, 2018-09-26:
**Type of issue**: bug report | feature request

**Impact**: no functional change

**Development Phase**: request

**Please tell us about your environment:** Chisel 3.1

**What is the use case for changing the behavior?**

Is there a way for us to suppress feature warnings (or at the very least "reflective access of structural type member" warnings) which are irrelevant to users?

Currently running sbt raises feature warnings to users by default:
```
[warn] there were 1 feature warnings; re-run with -feature for details
[warn] one warning found 
```

Enabling them leads sbt/scalac to complain excessively about accessing `io` fields, which is confusing to users since they have to wonder if this is a bug in their code (which it is not):

```
[warn] /.../src/main/scala/GCD.scala:32:6: reflective access of structural type member value outputGCD should be enabled
[warn] by making the implicit value scala.language.reflectiveCalls visible.
[warn]   io.outputGCD := x
[warn]      ^
[warn] /.../src/main/scala/GCD.scala:33:6: reflective access of structural type member value outputValid should be enabled
[warn] by making the implicit value scala.language.reflectiveCalls visible.
[warn]   io.outputValid := y === 0.U
[warn]      ^
[warn] 5 warnings found
[info] Done compiling.
[info] Compiling 1 Scala source to /.../target/scala-2.12/test-classes ...
[warn] /.../src/test/scala/GCDTester.scala:36:19: reflective access of structural type member value value1 should be enabled
[warn] by making the implicit value scala.language.reflectiveCalls visible.
[warn] This can be achieved by adding the import clause 'import scala.language.reflectiveCalls'
[warn] or by setting the compiler option -language:reflectiveCalls.
[warn] See the Scaladoc for value scala.language.reflectiveCalls for a discussion
[warn] why the feature should be explicitly enabled.
[warn]       poke(gcd.io.value1, i)
[warn]                   ^
[warn] /.../src/test/scala/GCDTester.scala:37:19: reflective access of structural type member value value2 should be enabled
[warn] by making the implicit value scala.language.reflectiveCalls visible.
[warn]       poke(gcd.io.value2, j)
```

edwardcwang, 2018-09-26:
Conclusion: add `scalacOptions += "-language:reflectiveCalls"` to the chisel template at some point, and maybe investigate why it's happening

jackkoenig, 2018-09-26:
Well the "why" is that standard Chisel style uses structural typing, eg.
```scala
val io = IO(new Bundle {
  val foo = Output(UInt(8.W))
})
```
accessing fields of structural types requires reflection, which is not available on all platforms (eg. ScalaJS)