chipsalliance/chisel #900: Make When Statement Functional

standbyme, 2018-09-30:
**Type of issue**: feature request

**Impact**: API modification

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
`when` statement can't return its result of last line like `if` in Scala.
For example
`if` in Scala
val a = if(2>1){
    6
}else{
    7
}
assert(a==6)
`when` in Chisel
val a = when(2>1){
    6
}else{
    7
}
// a is Unit there
**What is the expected behavior?**
val a = when(2.U>1.U){
    6
}else{
    7
}
assert(a==6)
**Please tell us about your environment:**

**What is the use case for changing the behavior?**
I'm writing Bitonic Sort so I need to split a Vec into min and max parts
```scala
    val half = n / 2
    val (a, b) = io.in.splitAt(half)
    val (min, max) = a
      .zip(b)
      .map {
        case (m, n) => {
          if ((m < n).asUInt == 1.U) { // it will not work. I don't know why this condition always false. I want to use `when` there but `when` return nothing
            (m, n)
          } else {
            (n, m)
          }
        }
      }
      .unzip
```

edwardcwang, 2018-09-30:
We can express the provided example as a Mux (with some modifications):

```scala
val a = Mux(2.U>1.U, 6.U, 7.U)
// a === 6.U
```

A related problem with the example is that it's trying to return a Scala Int (6) instead of a hardware UInt (`6.U`). When we use the `when` statement or Mux, we are constructing hardware, so it doesn't make sense to return a software constant as the output of a hardware mux.

standbyme, 2018-10-01:
@edwardcwang 
Thanks for your help. I use `Mux` instead of `if` in my Bitonic Sort:
```scala
Mux(m < n, (m, n), (n, m)) // it fails because Tuple can't be used in Mux
```
I have solved this problem with `Mux`:
```scala
val half = n / 2
val (a, b) = io.in.splitAt(half)
val min = (0 until half).map(i => Mux(a(i) < b(i), a(i), b(i)))
val max = (0 until half).map(i => Mux(a(i) > b(i), a(i), b(i)))
```
However, I think that it is a little miscellaneous. Is there a better way?

colinschmidt, 2018-10-01:
For tuple muxes see: https://github.com/freechipsproject/chisel3/issues/475

jackkoenig, 2018-10-01:
And https://github.com/freechipsproject/chisel3/issues/864 talks about HardwareTuple which I have proposed as the solution to Muxing tuples

standbyme, 2018-10-02:
Thanks for everyone's help!

closed 2018-10-02 by standbyme