chipsalliance/chisel #904: [RFC] Formalize support for IP-XACT

chick, 2018-10-08:
IP-XACT is a xml metadata format for describing interaces to IP circuit designs.

## Motivation
There is a lot of interest from CAD vendors for us to support IP-XACT to streamline integration with their tools.  A lot of work has been done here, particularly with respect to the Craft 2 project. It is not clear that this code has been kept current, or was ever adequately documented.

## Current state
IP-XACT implementations exist in 
- [rocket-dsp-utils](https://github.com/ucb-art/rocket-dsp-utils/blob/master/src/main/scala/ipxact/IPXact.scala)
- [rocket-dsp-utils](https://github.com/ucb-art/rocket-dsp-utils/blob/diplomacyPort/src/main/scala/dspblocks/IPXact.scala) and again here, with diplomacy support.
- There has also been work done by Intel [Reference Needed]()

## Impact
This should be minimal disruptive because it does not alter existing API's. There are several
steps to solidify support
- Identify the right repository for it to live in.
- Test existing code and modernize as necessary
- Provide more Documentation

edwardcwang, 2018-10-08:
Can it live as a separate library?

chick, 2018-10-08:
Yes, I think it can although it may require (or already have) hooks in to
the Builder State to help implement it.
But otherwise I think this would seem to be yet another
Annotation/Transform based library.
-chick

Chick Markley
Staff Programmer -- Adept Lab
University of California, Berkeley
583-2 Soda Hall, Berkeley, CA 94720-1776
chick@berkeley.edu

jackkoenig, 2018-10-08:
What kind of hooks are necessary? They might be useful for other libraries as well.

Nic30, 2018-10-12:
Hello, I have very similar tool, it has support also for Quartus *_hw.tcl format.
I would like to kill it in favor of chisel3 ecosystem. I can help you with you implementation if you want.

grebe, 2018-10-12:
One issue with ipxact is that the ipxact library we use does not seem to live in maven- perhaps I simply haven't searched hard enough. That's an annoyance for build and deployment.

The approach in rocket-dsp-utils has some issues, the biggest of which is that it generates ipxact from the top level. If someone doesn't use a dspblock, they don't get ipxact out.

I think a better approach (that can hopefully use a lot of the same underlying code) is to have an API for annotating bundles with their parameterization. For example, an AXI-4 bundle could be annonated saying it is a slave with such-and-such addresses. A separate API for annotating modules to describe their own parameterization would also be good. This API should probably live in chisel.

A seperate firrtl pass can extract these parameters, along with structural information about widths, to generate ipxact (in principle, you could generate other formats too- say for xilinx ip integrator). This pass doesn't have to live in firrtl or chisel, but perhaps it makes sense to keep it in chisel anyways. It would be overkill to give it its own repo.

Nic30, 2018-10-12:
I think that It is good to have IP-core informations associated with definition of interface because of consystency.
And it is bad to have IP-core generator in chisel because it is large and unrelated code.

In my framework I have api for annotations and 
https://github.com/Nic30/hwtLib/blob/master/hwtLib/amba/axis.py#L71

And ip core generator is separated.
[ipCorePackager](https://github.com/Nic30/ipCorePackager)