chipsalliance/chisel #924: sbt compile error

MikuChaser, 2018-11-03:
OS: Ubuntu 16.04
Problem: Can't use chisel 3.2+
Description: I have installed the newest version of sbt, firrtl and verilator. When I created my project under the folder  chisel-template or chisel-tutorial, the project can work. However, using the chisel3 repository is wrong. When I run "sbt compile", it shows:
`[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/wubin/chisel-workspace/chisel3/project
[info] Loading settings from build.sbt ...
[info] Set current project to chisel3 (in build file:/home/wubin/chisel-workspace/chisel3/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Updating coreMacros...
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT: java.text.ParseException: [unknown tag project in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag modelVersion in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag groupId in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag artifactId in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag packaging in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag description in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag version in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag organization in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag scm in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag connection in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag developerConnection in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag licenses in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag license in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag distribution in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag developers in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag developer in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag id in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[warn] ]
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:firrtl_2.12:1.2-SNAPSHOT
[warn] 		  +- edu.berkeley.cs:coremacros_2.12:3.2-SNAPSHOT
[error] sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT: java.text.ParseException: [unknown tag project in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag modelVersion in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag groupId in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag artifactId in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag packaging in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag description in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag version in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag organization in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag scm in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag connection in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag developerConnection in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag licenses in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag license in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag distribution in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag developers in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag developer in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag id in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] ]
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:331)
[... 50 lines cut ...]
[error] , unknown tag packaging in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag description in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag version in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag organization in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag scm in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag connection in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[... cut ...]
[error] , unknown tag developer in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag id in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag name in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] , unknown tag url in file:/home/wubin/.ivy2/cache/edu.berkeley.cs/firrtl_2.12/ivy-1.2-SNAPSHOT.xml.original
[error] ]
[error] Total time: 6 s, completed Nov 3, 2018 2:00:17 PM
`
What should I do?

seldridge, 2018-11-03:
If you're trying to build Chisel locally (which is generally unnecessary unless you're doing development on the repo), you need to have published a local snapshot of the FIRRTL repository. Inside the FIRRTL reposotiory run `sbt publishLocal`. This puts all the FIRRTL jars in ~/.ivy2.

The reason why chisel-template works is that that is pulling directly from Maven. The Chisel repository is configured to use your local snapshot of FIRRTL as this makes the most sense for developer work that would be contributed back upstream.

MikuChaser, 2018-11-03:
> If you're trying to build Chisel locally (which is generally unnecessary unless you're doing development on the repo), you need to have published a local snapshot of the FIRRTL repository. Inside [...]

Of course. According to the readme of the Chisel3, I have installed the firrtl.
[image]
Firstly, I git from the repo. Then, run "sbt compile" and it's successful. But "sbt test" failed. "sbt assembly" and "sbt publishLocal" also work.

MikuChaser, 2018-11-05:
I have solved it. Just delete the old .ivy2 folder.

seldridge, 2018-11-05:
Closing this as it looks like you have a solution. Feel free to re-open if that's not the case.

closed 2018-11-05 by seldridge