chipsalliance/chisel #929: [Bug] Wrong pruning behavior when templates are used in Chisel3 or FIRRTL?

vegaluisjose, 2018-11-07:
**Type of issue**: bug report

**Impact**: unknown

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

One operand is being removed when template is used. Take the following example, when I inspect the `.fir` file, I saw that `b` port was optimized out from module `Two` and `One`. This happens in a more complex project that I am working on and this is the simplest example I could come up with that reproduces the behavior. If I don't use templates, then the problem is gone.

```
package project.test

import chisel3._
import chisel3.util._

class OneType[T <: Data](dat_T: T) extends Bundle {
  val a = dat_T
  val b = dat_T
  override def cloneType = new OneType(dat_T).asInstanceOf[this.type]
}

class TwoType[T <: Data](dat_T: T) extends OneType(dat_T) {
  val en = Bool()
  override def cloneType = new TwoType(dat_T).asInstanceOf[this.type]
}

class Ctrl extends Module {
  val io = IO(new Bundle {
    val a = Output(UInt(8.W))
    val b = Output(UInt(8.W))
  })
  io.a := 8.U
  io.b := 7.U
}

class One extends Module {
  val io = IO(new Bundle {
    val op = Input(new OneType(dat_T = SInt(8.W)))
  })
  printf("a:%x b:%x\n", io.op.a, io.op.b)
}

class Two extends Module {
  val io = IO(new Bundle {
    val clt = Input(new TwoType(dat_T = SInt(8.W)))
  })

  val one = Module(new One)

  one.io.op.a := io.clt.a
  one.io.op.b := io.clt.b
}

class Test extends Module {
  val io = IO(new Bundle {
    val d = Input(Bool())
  })

  val ctr = Module(new Ctrl)
  val dut = Module(new Two)

  dut.io.clt.en := false.B
  dut.io.clt.en := false.B
  dut.io.clt.a := ctr.io.a.asSInt
  dut.io.clt.b := ctr.io.b.asSInt
}

object Elaborate extends App {
  chisel3.Driver.execute(args, () => new Test)
}
```

The FIR file

```
buildInfoPackage: chisel3, version: 3.1.3, scalaVersion: 2.11.12, sbtVersion: 1.1.1, builtAtString: 2018-09-12 19:37:29.007, builtAtMillis: 1536781049007
circuit Test :
  module Ctrl :
    input clock : Clock
    input reset : UInt<1>
    output io : {a : UInt<8>, b : UInt<8>}

    io.a <= UInt<4>("h08") @[Test.scala 22:8]
    io.b <= UInt<3>("h07") @[Test.scala 23:8]

  module One :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip op : {a : SInt<8>}}

    node _T_9 = bits(reset, 0, 0) @[Test.scala 30:9]
    node _T_11 = eq(_T_9, UInt<1>("h00")) @[Test.scala 30:9]
    when _T_11 : @[Test.scala 30:9]
      printf(clock, UInt<1>(1), "a:%x b:%x\n", io.op.a, io.op.a) @[Test.scala 30:9]
      skip @[Test.scala 30:9]

  module Two :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip clt : {a : SInt<8>, en : UInt<1>}}

    inst one of One @[Test.scala 38:19]
    one.clock <= clock
    one.reset <= reset
    one.io.op.a <= io.clt.a @[Test.scala 40:15]
    one.io.op.a <= io.clt.a @[Test.scala 41:15]

  module Test :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip d : UInt<1>}

    inst ctr of Ctrl @[Test.scala 49:19]
    ctr.clock <= clock
    ctr.reset <= reset
    inst dut of Two @[Test.scala 50:19]
    dut.clock <= clock
    dut.reset <= reset
    dut.io.clt.en <= UInt<1>("h00") @[Test.scala 52:17]
    dut.io.clt.en <= UInt<1>("h00") @[Test.scala 53:17]
    node _T_9 = asSInt(ctr.io.a) @[Test.scala 54:28]
    dut.io.clt.a <= _T_9 @[Test.scala 54:16]
    node _T_10 = asSInt(ctr.io.b) @[Test.scala 55:28]
    dut.io.clt.a <= _T_10 @[Test.scala 55:16]
```

**What is the current behavior?**
```
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
a:07 b:07
```
**What is the expected behavior?**
```
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
a:08 b:07
```
**Please tell us about your environment:**

- Scala version: 2.11.12
- sbt version: 1.1.1
- Chisel3 version: 3.1.3
- FIRRTL version: 1.1.3

seldridge, 2018-11-07:
This is interesting... If you make `dat_T` call-by-name, then this works:
```scala
class OneType[T <: Data](dat_T: => T) extends Bundle {
  val a = dat_T
  val b = dat_T
  override def cloneType = new OneType(dat_T).asInstanceOf[this.type]
}
```

edwardcwang, 2018-11-07:
Couldn't `chiselTypeOf` be used here?

vegaluisjose, 2018-11-08:
@seldridge I see, calling by name in both types `OneType` and `TwoType` fixes the issue.

@edwardcwang is there an example on using `chiselTypeOf`? what would be the advantage?

Thanks both of you,

jackkoenig, 2018-12-18:
subsumed by https://github.com/freechipsproject/chisel3/issues/765

closed 2018-12-18 by jackkoenig