chipsalliance/chisel #932: PeekPokeTester doesn't work with StrongEnum

nhynes, 2018-11-10:
`poke`:

```scala
[error] StateMachineTest.scala:42:00: overloaded method value poke with alternatives:
[error]   (signal: chisel3.core.Aggregate,value: IndexedSeq[BigInt])Unit <and>
[error]   (signal: chisel3.Bundle,map: Map[String,BigInt])Unit <and>
[error]   (signal: chisel3.Bits,value: Long)Unit <and>
[error]   (signal: chisel3.Bits,value: Int)Unit <and>
[error]   (signal: chisel3.Bits,value: BigInt)Unit <and>
[error]   (path: String,value: Long)Unit <and>
[error]   (path: String,value: Int)Unit <and>
[error]   (path: String,value: BigInt)Unit
[error]  cannot be applied to (MyStrongEnum.Type, MyStrongEnum.Type)
[error]   poke(dut.io.state, MyStrongEnum.Idle)
```

`expect`:

```scala
[error] StateMachineTest.scala:42:00: overloaded method value expect with alternatives:
[error]   (signal: chisel3.Bundle,expected: Map[String,BigInt])Boolean <and>
[error]   (signal: chisel3.core.Aggregate,expected: IndexedSeq[BigInt])Boolean <and>
[error]   (signal: chisel3.Bits,expected: BigInt,msg: => String)Boolean <and>
[error]   (good: Boolean,msg: => String)Boolean
[error]  cannot be applied to (MyStrongEnum.Type, MyStrongEnum.Type)
[error]   expect(dut.io.state, MyStrongEnum.Idle)
```

seldridge, 2018-11-10:
I haven't looked at this, but is converting to `UInt` for the `poke` or `expect` with `.asUInt` an option / workaround?

hngenc, 2018-11-10:
I'll take a look at this soon, but in the meantime, I suspect that `poke(dut.io.state, MyStrongEnum.Idle.litValue())` and `expect(dut.io.state, MyStrongEnum.Idle.litValue())` might work.

nhynes, 2018-11-11:
> `poke` using `.U` or `expect` with `.asUInt`

@seldridge That would have been a nice fix, but that produces the following error:
```scala
[info]   chisel3.internal.ChiselException: Error: Not in a UserModule.
Likely cause: Missed Module() wrap, bare chisel API call, or attempting to
construct hardware inside a BlackBox.
[info]   at chisel3.internal.throwException$.apply(Error.scala:13)
[info]   at chisel3.internal.Builder$.forcedUserModule(Builder.scala:227)
[info]   at chisel3.internal.Builder$.pushOp(Builder.scala:259)
[info]   at chisel3.core.EnumType.do_asUInt(StrongEnum.scala:73)
```

I guess a workaround workaround would be to use a `BasicTester` which _would_ allow constructing hardware, but that would make peeking and poking a bit more difficult.

hngenc, 2018-11-11:
I think I can get a fix for this out in the next day or two. I'm not sure I understand what you mean by the enum's internal UInt field; enums aren't composed of other Chisel objects, although they do get translated to UInts when going from Chisel to FIRRTL.

nhynes, 2018-11-11:
[This is what I meant by "interal UInt"](https://github.com/freechipsproject/chisel3/blob/4dddde56d0c1a66a05457c7dda409238d2de8e8a/chiselFrontend/src/main/scala/chisel3/core/StrongEnum.scala#L157), but I also don't have a great mental model of what happens during elaboration.

Thanks for the fix!

hngenc, 2018-11-12:
Actually, looking at this problem again, I'm unsure which of the possible solutions to implement. None of them seem ideal to me.

The problem is that `peek`, `poke`, `pokeAt`, etc. all expect inputs of type `Bits`, but enums aren't subclassed from `Bits`.

1. We can make enums inherit from `Bits` instead of inheriting directly from `Element`. I don't like this solution, because `Bits` have operations like `<<` and `toBool` that I don't think make sense for enumerations.
2. We can change `peek`, `poke`, etc. to accept arguments of type `Element`. However, users will then be able to poke `Analog`, `Clock`, etc. and I don't know if we want to permit that.
3. We can overload `poke`, `peek`, etc. to accept arguments of type `EnumType` as well. I don't like this solution because there will be some code duplication. Also, functions like `pokeAt[TT <: Bits]` can't be differentiated from `pokeAt[TT <: EnumType]` after type erasure, so the overloading may get a little hacky at some points.

If nobody has any preferences, then I guess I'll just make enums inherit from `Bits`. I can just overload the `<<` and `toBool` functions to throw exceptions at elaboration time.

nhynes, 2018-11-12:
Proposal 2 sounds reasonable if `Pokeable` subsumes conversion to an `expect`able type. Proposal 4 would also be fine if using a macro to generate the impls for both; this is the Rust/C++-y way.

edwardcwang, 2018-11-12:
Building off Nick's comment - could we add a `Pokeable` trait that Bits, Element, etc can mix in?

hngenc, 2018-11-12:
A conversion from `EnumType` to `UInt` would be great, but I can't figure out how to do it outside of a `Module`. Maybe somebody who's more familiar with the Chisel codebase will know, if its even possible.

@edwardcwang I considered that as well. As I understand it, we'd have to mix in the trait in `chisel3`, but it wouldn't have any meaning except in `chisel-testers` which is a different project. That feels like a coding sin to me, but it would be a simple solution.

hngenc, 2018-11-14:
@nhynes If you need enum testing support right away, then you can pull this branch: https://github.com/hngenc/chisel-testers/tree/add_enum_support. I haven't tested it extensively, but it seems to work for examples like `poke(dut.io.state, MyStrongEnum.Idle)`. If the reviewers are fine with my changes, then I should be able to upstream this soon.

closed 2018-12-18 by chick