chipsalliance/chisel #933: How to connect Mem with wire?

myuanqing, 2018-11-10:
I feel the must to connect memory with wire.

In verilog, we may write like this:
**Reg [15:0]mem[25:0];
Wire[15:0]w[25:0];
w = mem** 

 However, I find that there is no such method for type Mem. 

How can I solve this problem?

//I'm a foreigner. My English is poor......Sorry for that.

edwardcwang, 2018-11-10:
So it seems like it's just getting read out directly from the reg in Verilog. Maybe the synthesis tool is doing something funny?

```verilog
  reg [25:0] tag [0:15]; // @[cmd3.sc 15:18:@8.4]
  // ...
  wire [25:0] tag__T_77_data; // @[cmd3.sc 15:18:@8.4]
  wire [3:0] tag__T_77_addr; // @[cmd3.sc 15:18:@8.4]
  // ...
  assign tag__T_77_addr = 4'h4;
  assign tag__T_77_data = tag[tag__T_77_addr]; // @[cmd3.sc 15:18:@8.4]
  // ...
  assign io_tag_4 = tag__T_77_data;
```

myuanqing, 2018-11-11:
@edwardcwang I rewrite the verilog file like this
 
[image]

I get the the schematic:
[image]

io_tags now directly connect to tag_regs. 
More Beautiful， isn't it?

Is there anyway to get this directly in chisel:
 **assign tag__T_77_data = tag[4'h4];** 

PS:
I use Vivado to do the synthesis job. Maybe another tool would do better?

myuanqing, 2018-11-11:
@edwardcwang  Thank you for your answer   ：D

edwardcwang, 2018-11-11:
My point is that if you read the generated Verilog as I showed above, it *is* the same as `assign io_tag_4 = tag[4'h4];`.

myuanqing, 2018-11-11:
@edwardcwang But the schematic of `assign io_tag_4 = tag[4'h4]` and the schematic of `assign tag__T_77 = 4'h4`;

 `assign io_tag_4 = tag[tag__T_77]` do differ! 
The logic synthesis tool doesn't simplify the mux.

edwardcwang, 2018-11-11:
Hmm, weird. Try looking at the post-synthesis netlist? Usually most synthesis tools will eliminate such a mux.

myuanqing, 2018-11-11:
@edwardcwang  Thank you! It does eliminate the mux! 👍 👍 👍

myuanqing, 2018-11-11:
solved!

closed 2018-11-11 by myuanqing