chipsalliance/chisel #934: Custom always block

bhimsenpadalkar, 2018-11-13:
**Type of issue**: feature request | other enhancement

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information**
 Currently, chisel generates "always" block by default for implicit clock signal and logic inside "when" block goes inside it.
Can we add our custom always block independent of the implicit one which will allow me build the logic around rising/falling edge of any signal?

seldridge, 2018-11-13:
Multiple clock domains (with everything posedge triggered) are available via the `withClock` and `withClockAndReset` APIs. Some documentation is available on the [wiki](https://github.com/freechipsproject/chisel3/wiki/Multiple-Clock-Domains) and in the [MultiClockSpec tests](https://github.com/freechipsproject/chisel3/blob/master/src/test/scala/chiselTests/MultiClockSpec.scala) about how you can use this. 

Do those suit your use cases?

If you're asking specifically about negedge and pos/negedge triggered things, that's a more complicated endeavor. This was discussed a bit in https://github.com/freechipsproject/firrtl/issues/695 and https://github.com/freechipsproject/chisel3/issues/736, but was stalled looking for concrete use cases. Most of the expected use cases are some type of hard IP or one-time designed Verilog blackboxes. Additionally, there was also some hesitation on allowing users to use negedge triggered things haphazardly. That's always seemed to be, outside of some specific use cases, an anti-pattern for hardware design. Consequently, there wasn't a big push towards supporting negedge and pos/negedge, though I'm sure that will resurface at some point.

bhimsenpadalkar, 2018-11-13:
withClock worked perfectly fine for my use case.. Thanks!!

closed 2018-11-13 by bhimsenpadalkar