chipsalliance/chisel #940: Help with design low-level HDL language

XVilka, 2018-11-16:
FPGA world suffers a lot from fragmentation - some tools produce Verilog, some VHDL, some - only subsets of them, creating low-level LLVM-like alternative will help everyone, so HDL implementations will opt only for generating this low-level HDL and routing/synthesizers accept it. LLVM or WebAssembly - you can see how many languages and targets are supported now by both. With more open source tools for FPGA this is more feasible now than ever.

See SymbiFlow/ideas#19

seldridge, 2018-11-16:
Are you aware of [FIRRTL](https://GitHub.com/freechipsproject/FIRRTL)? Chisel3 already compiles to FIRRTL (a flexible intermediate representation of RTL analogous to "LLVM IR for circuits"). The FIRRTL compiler is an optimizing compiler with a provided backend for Verilog. Any FPGA/ASIC tool vendor, if they so desire, can directly use FIRRTL for synthesis and place and route if they want to support it. Any user wanted to build a new frontends can target FIRRTL. Any user wanting to write a new backend (e.g., for VHDL) can consume FIRRTL.

FIRRTL is battle tested, having been used to generate Verilog used for numerous tape outs at Berkeley, SiFive, Google, and other institutions/companies

seldridge, 2018-12-18:
Closing this in favor of the discussion on https://github.com/SymbiFlow/ideas/issues/19. We're open to RFCs on the broader Chisel/FIRRTL ecosystem on how to best handle this and align FIRRTL with the needs of other frontend developers: https://github.com/freechipsproject/rfcs.

closed 2018-12-18 by seldridge