chipsalliance/chisel #944: Bits is a second class type

schoeberl, 2018-11-17:
**Type of issue**: bug report

On trying to port the Patmos code base to Chisel 3 I realize that type Bits is not anymore a full citizen of the Chisel type system.

I know that there was a discussion on how to handle Bits. I personal think that Bits as additional type, which has no arithmetic meaning, makes sens. Consider a bus: a bus type should not support addition. Chisel is (hopefully) also about type safety. Some some hardware should simply not be represented as an int.

Take as an example VHDL. It has a std_logic_vector type where you cannot perform arithmetic operations.

If you agree, I can give it a shot to add the missing methods to Bits to make it a usable type, and *most important* make it compatible to Chisel 2.

**Impact**: API modification

quite some rewrite on the Patmos code base.

edwardcwang, 2018-11-17:
For buses, I agree it doesn't make sense to add buses. At the same time it seems more logical to use a `Bundle` (or Record) for them, no? A `Bundle` has more semantic meaning than a collection of bits.

schoeberl, 2018-11-17:
Sure a bundle for a bus, but a bundle of what? Address, data, control signals should all be Bits in my opinion.

One more argument: Trying to compile the Patmos source with Chisel gave me almost 700 errors. And many of them are related to operation on Bits that are not supported anymore. If this issue is for our project, it could be also a show stopper for others trying to port from Chisel2 to 3.

Cheers,
Martin

colinschmidt, 2018-11-19:
I would look at https://github.com/freechipsproject/chisel3/wiki/Chisel3-vs-Chisel2 when porting code. And if there is something missing there it would be great to add to it. Its not often someone does this porting task so its hard to get real feedback on that page.

chick, 2018-11-20:
@schoeberl Where can one find the PATMOS chisel code?

schoeberl, 2018-11-22:
at https://github.com/t-crest/patmos <https://github.com/t-crest/patmos> within hardware/src/main...

You can reproduce the Patmos Chisel 3 errors by changing Chisel libraries from 2 to 3 in patmos/hardware/build.sbt and then just compile with sbt from within patmos/hardware

sbt "runMain patmos.PatmosMain"

Cheers,
Martin

ducky64, 2018-12-15:
Superseded by #957

closed 2018-12-15 by ducky64